supermerill/SuperSlicer #3386: Nozzle hit part in sequential print

Aimen06, 2022-10-29:
### What happened?

The nozzle hits the part when printing in sequential.

### Project file & How to reproduce

I duplicate the part with new instance button.
I press many time to have enough parts.
When i press the arrange to arrange all the part.
I have notice that the part are not one after the other.
The first is after the 6 etc...

[VisV5.3mf.zip](https://github.com/supermerill/SuperSlicer/files/9894563/VisV5.3mf.zip)
[image]

### Version

2.4

### Operating system

mac os  12.5

### Printer model

mingda magician max

jhlchu, 2022-10-30:
You don't need to press `Add Instance` every time. Use `Set number of instances`.

Aimen06, 2022-10-30:
It's the same thing