supermerill/SuperSlicer #3390: Blacklist specific speeds to avoid resonance

tonyshackelton, 2022-10-31:
I'm proposing a "speed blacklist", 1 range (i.e. 50-60mm/s). When set, the slicer would either speed those moves in that range up or down to avoid the resonant frequency of the printer. This is would be extremely nice for core XY machines running spreadcycle. I'm sure I can't be the only one that fiddles with all the speed settings and layer times to avoid hitting "that one speed" where your printer sounds like nails on a chalkboard.

diggit, 2022-11-02:
I don't think, that slicer is right part of printing chain where to mitigate this issue. You should rather find and fix that issue root cause.
Quite often, I can see stepper drivers capable of several amps used at much lower current (eg 1.5 A driver and configured for 0.5 A). In such case, TMC driver current resolution degrades and become noisier due to that. It can be fixed/improved by changing current sense resistors.

tonyshackelton, 2022-11-03:
I'm not talking about fixing a badly mis-configured machine but, no printer will be happy at all speeds. Since slicer controls print speed, so I think it's perfectly logical to request it to avoid certain speeds. It has speed min's and max's already, and a large part of the app is devoted to speed control, even edge cases like overhangs, perimeters, etc.. Retuning the firmware for this seems extreme, although "custom g-code" that tweaks the TMC drivers based on a print profile has merit, and input shaping helps here too.

tonyshackelton, 2022-11-03:
Maybe some context would be helpful... I have a pretty nice vzbot, big steppers, tuned drivers, input shaper, etc... I'm able to really crank it up way past anything I could flow or cool. So I'll slice up a model, make sure the flow is "up there" but not "too much". Printing great, woosh woosh, hauling.... then towards the top of the model (just about any model), the layer time governor starts to slow things down so they have time to cool. As it progressively slows and slows it will always cross over that speed that causes resonant vibrations (no matter what the speed happens to be). One moment, woosh woosh, then out of nowhere, dogs start howling, you wife asks "what's that racket?", then a min later it's gone again... I sure would like an easy way to avoid that. I can't think of an easier way than just telling superslicer.

I use auto-cooling in the example, but it would apply to any of the automatic speed reducing features. It's just prominent and unavoidable in this case. 

Sort of just disabling those features outright, I'd love to hear any ideas to avoid this situation while slicing.

diggit, 2022-11-05:
Did you analyze what is resonating? Typical issue is mechanical resonance of steppers.
> ...big steppers...

Yeah, the bigger the stepper is, the lower the resonance frequency is (due to higher rotor inertia). At least from common sense...

> ...tuned drivers...
Can you elaborate on that?
I know it's basic thing but did you try to play with micro-stepping? Does it influence intensity of resonance?

I am asking because it makes sense to mitigate issue in higher levels once you run out of options in lower levels.

tonyshackelton, 2022-11-06:
I'm running 128 steps right now, but with interpolation flipped on it's basically "upscaled" to 256 regardless. I ran default chopper settings for a bit, then went to ratrig's config for this motor driver/pair (2209/2504), most recently, I downloaded the TMC chopper spreadsheet and worked out the math on that. It helped a lil... Like you implied above, big steppers are a trade off. I can certainly move around the offending frequency/sound with these settings, but it's always there... and it's not like even it's causing print artifacts, It's just annoying. Finer control over speeds would be nice and I figured it would be very easy to implement, hence the feature request. 

Here are my current settings, since you asked (no pun intended :) )
run_current: 1.6 <-- max I can run on these 2209's with cooling
interpolate: True <-- basically upscale to 256 microsteps, but helps with hold sounds
stealthchop_threshold: 0 <-- disable stealthchop which silent when slow but problematic at speed
--settings per TMC chopper spreadsheet https://www.trinamic.com/fileadmin/assets/Products/ICs_Documents/TMC2209_Calculations.xlsx--
driver_TBL: 1 
driver_TOFF: 3
driver_HSTRT: 7 
driver_HEND: 8

This is starting to remind me of an experience I had with a VW microbus. Those old transmissions were LOUD... when I asked the local VW shop owner what I could do, he pointed over at the car stereo display in the corner. :) Moral of story: I just need to finish my enclosure.

misiek271, 2022-11-07:
you have it on klipper? do you make resonance calibration in klipper using adxl accelerometer. it helps AAA LOOOT for me it was accel 1.5k with no resonance and after calibration its a 7k.
resonance is a bit:::.* when you move your printer to a different ground like shelf or floor resonance is changing , even belt tension change resonance of printer.

diggit, 2022-11-12:
@tonyshackelton I was thinking about it a little bit more. You need to avid specific **motor** speeds and not motion speed. It depends on printer motion system, which is unknown to slicer. It would make more sense to have such feature in printer SW/FW itself (eg. klipper.).

Lecso11, 2022-12-01:
I'd like that too. The steppers self frequency is something you cannot fix. We need to avoid that speed range. For me it is 42-55mm/s.