supermerill/SuperSlicer #3393: 2.5.59.0 "Failed to load image" loop message crash

Fuechsl, 2022-11-01:
### Version
2.5.59.0

### Operating system type + version
MacOS 13.0

### Behavior
On loading a project sometimes I get a message "Failed to load image" which loops infinitely (see attached screenshot). When I shut down ss and restart it, it works.

[image]

#### Project File (.3MF) where problem occurs
[Prj Füllbecher.3mf.zip](https://github.com/supermerill/SuperSlicer/files/9908112/Prj.Fullbecher.3mf.zip)
The issue occurs with different files, this is one of them. It also occurs somtimes, when I open a gcode file in the gcode viewer.

Thanks!

allenrowand, 2022-12-23:
Same issue here. After force quitting and reopening a couple times it will launch and display the bed PNG.

Fuechsl, 2024-07-16:
Having this issue AGAIN with 2.5.60.0. Superslicer crashes every time when started.

[image]