supermerill/SuperSlicer #3394: Superslicer 2.5.59.0 Arachne generator gaps in perimeters/walls

kot0005, 2022-11-02:
### What happened?

i am getting gaps in-between walls/perimeters , more pronounced near angle'd sections

### Project file & How to reproduce

Superslicer 2.5.59.0 Arachne generator :

[image]
[image]

This image is using cura 5.2.1 with arachne 

[image]

i tried a few settings like wall thickness, top flow rate etc but nothing seemed to change

### Version

2.5.59.0

### Operating system

windows 10

neophyl, 2022-11-02:
Yep, PS does this too with some models.  Its why I think the PS implementation isn't ready to be made the default like they have done.

diggit, 2022-11-03:
It looks like typical pressure advance configuration issue. Too high pressure advance causes such gaps in corners.
Or is it also visible in gcode?

CCS86, 2022-11-04:
Checking the grams of filament used when slicing with classic vs arachne perimeters generally shows a significant reduction with arachne. This seems to corroborate the under-extrusion people are observing.

kot0005, 2022-11-05:
> Checking the grams of filament used when slicing with classic vs arachne perimeters generally shows a significant reduction with arachne. This seems to corroborate the under-extrusion people are observing.

oh what, i will turn it off and try

kot0005, 2022-11-12:
those settings dont work, just using standard super slicer for now for now