supermerill/SuperSlicer #3398: Acceleation setting for first layer only after starting a line

user1419, 2022-11-03:
**Is your feature request related to a problem? Please describe.**
If there are adhesion problems, they often start at the beginning of a "line" when the extrusion first starts (the first few mm are getting dragged behind the nozzle before starting to stick). I found that a good solution for that problem is to lower the acceleration for the whole first layer, which takes a lot of time...

**Describe the solution you'd like**
If it would be possible to set the acceleration for the start of each line separately, that would save a lot of time while still solving the problem of the start of the extrusion getting dragged with the nozzle.

**Describe how it would work**
Maybe that feature could be added in post? Just detect whenever the Z height changes in the negative direction to the first layer height and lower the acceleration for a short moment, before raising it to it's previous value a few mm of extrusion later?

**Describe alternatives you've considered**
-

**Additional context**
Maybe there could even be one more setting for lowering the acceleration in sharp corners aswell, without affecting the whole first layer.

NiiXXiiN, 2023-10-29:
I was thinking about first layer accelerations and i believe it would be better to separate first layer accelerations into perimeter and infill. We lose so much time with low acc infill. It would be an amazing time saver. Perimeters low and infill max acc.

supermerill, 2023-11-22:
Also a "don't reverse-wipe on first layer", i guess.