supermerill/SuperSlicer #3404: Wipe tower with absolute coordinates

D3bugg3RR, 2022-11-05:
I have an artillery sidewinder X2 and both "M600" from gcode and "pause print" from lcd sets the extruder on absolute coordinates.
The purge tower can be created only using extruder relative coordinates.
The M600 can be corrected in custom gcode adding M83 as it is a scheduled event, but a "pause print" can be everywhere and it is decided by the user, after the resume print the LCD sets the printer on M82 (extruder absolute coordinates) but there isn't any way to send an M83.

Possibility to generate the wipe tower with both relative and absolute extruder coordinates

I've tried to see if it's possible to modify the LCD firmware but I haven't found anything in the config file.

neophyl, 2022-11-06:
I’d be complaining about the firmware, as that’s not supposed to change from relative to absolute and vice versa, it’s supposed to restore what it was before. 

You need to post this out to Artillery on whatever forums/contact methods they use so that they improve their code as I bet it’s effecting others. 

You could also try running one of the stock marlin firmwares instead but that’s a bit more technical. I ended up doing that on a creality to get around some of their issues.

jhlchu, 2022-11-06:
You can change extrusion coordinate for wipe tower with `Printer Settings > Custom G-Code > Between extrusion role change G-code`. I think if you do
```
{if extrusion_role=~/WipeTower/}
M83
{elsif last_extrusion_role=~/WipeTower/}
M82
{endif}
```
or something like that, you can switch back and forth between extrusion coordinates just for doing wipe towers.

D3bugg3RR, 2022-11-09:
> I’d be complaining about the firmware, as that’s not supposed to change from relative to absolute and vice versa, it’s supposed to restore what it was before.
>
> You need to post this out to Artillery [...]

\

> You can change extrusion coordinate for wipe tower with `Printer Settings > Custom G-Code > Between extrusion role change G-code`. I think if you do
>
> ```
> {if extrusion_role=~/WipeTower/}
> M83
> {elsif [...]

Thank you for the suggestion, I will try asap!

jhlchu, 2022-11-10:
If it works out, post an update and close the issue.