supermerill/SuperSlicer #3405: Unnecessary Solid Infill

ssjbardock, 2022-11-05:
### What happened?

Solid infill is added in random areas instead of standard infill, greatly increasing filament usage and time.

Seems to occur with pretty much any model that has smaller cross\-section areas.

### Project file & How to reproduce

[solid ifill problem.zip](https://github.com/supermerill/SuperSlicer/files/9944390/solid.ifill.problem.zip)

### Version

2.5.59.0

### Operating system

Windows 10

### Printer model

Custom

jhlchu, 2022-11-06:
`Print Settings > Infill > Advanced > Solid infill threshold area` I used `5mm^2` and it used sparse infill instead. Close if solved issue.

ssjbardock, 2022-11-06:
Yup that'll do it. Thanks!

closed 2022-11-06 by ssjbardock