supermerill/SuperSlicer #3406: crash on launch

jrpmedia, 2022-11-05:
### What happened?

crashes on launch with apple error log

### Project file & How to reproduce

launch app

-------------------------------------
Translated Report (Full Report Below)
-------------------------------------

Process:               SuperSlicer [97321]
Path:                  /Applications/SuperSlicer.app/Contents/MacOS/SuperSlicer
Identifier:            com.merill.slic3r/
Version:               SuperSlicer SuperSlicer_2.4.58.5_2022-09-23 (SuperSlicer_2.4.58.5_2022-09-23)
Code Type:             X86-64 (Native)
Parent Process:        launchd [1]
User ID:               501

Date/Time:             2022-11-05 20:01:11.1420 +0000
OS Version:            macOS 12.6.1 (21G217)
Report Version:        12
Bridge OS Version:     7.0 (20P411)
Anonymous UUID:        24D4E3CE-DDDC-8583-70AA-FAABBF2D3BDF

Sleep/Wake UUID:       8BED7075-FB19-4BCB-93A3-E0CACEBD37CA

Time Awake Since Boot: 83000 seconds
Time Since Wake:       2285 seconds

System Integrity Protection: enabled

Crashed Thread:        0

Exception Type:        EXC_CRASH (SIGABRT)
Exception Codes:       0x0000000000000000, 0x0000000000000000
Exception Note:        EXC_CORPSE_NOTIFY

Termination Reason:    Namespace DYLD, Code 1 Library missing
Library not loaded: '/usr/local/opt/zstd/lib/libzstd.1.dylib'
Referenced from: '/Applications/SuperSlicer.app/Contents/MacOS/SuperSlicer'
Reason: tried: '/usr/local/opt/zstd/lib/libzstd.1.dylib' (no such file), '' (no such file), '' (no such file)
(terminated at launch; ignore backtrace)

Thread 0 Crashed:
0   dyld                          	       0x10cdac98e __abort_with_payload + 10
1   dyld                          	       0x10cdc2a72 abort_with_payload_wrapper_internal + 80
2   dyld                          	       0x10cdc2aa4 abort_with_payload + 9
3   dyld                          	       0x10cd70699 dyld4::halt(char const*) + 615
4   dyld                          	       0x10cd6c5b5 dyld4::prepare(dyld4::APIs&, dyld3::MachOAnalyzer const*) + 3995
5   dyld                          	       0x10cd6b4e4 start + 388

Thread 0 crashed with X86 Thread State (64-bit):
  rax: 0x0000000002000209  rbx: 0x0000000000000000  rcx: 0x00007ff7bde51638  rdx: 0x00007ff7bde51aa0
  rdi: 0x0000000000000006  rsi: 0x0000000000000001  rbp: 0x00007ff7bde51680  rsp: 0x00007ff7bde51638
   r8: 0x00007ff7bde516a0   r9: 0x0000000000000000  r10: 0x0000000000000075  r11: 0x0000000000000246
  r12: 0x0000000000000075  r13: 0x00007ff7bde51aa0  r14: 0x0000000000000001  r15: 0x0000000000000006
  rip: 0x000000010cdac98e  rfl: 0x0000000000000246  cr2: 0x000000010cd72da4
  
Logical CPU:     0
Error Code:      0x02000209 
Trap Number:     133

Binary Images:
       0x10cd66000 -        0x10cdd1fff dyld (*) <7b87a986-a153-33c4-8470-d56410b7f9d5> /usr/lib/dyld

External Modification Summary:
  Calls made by other processes targeting this process:
    task_for_pid: 0
[... 100 lines cut ...]

      },
      "deploymentId" : 240000021
    },
    {
      "rolloutId" : "60186475825c62000ccf5450",
      "factorPackIds" : {

      },
      "deploymentId" : 240000026
    }
  ],
  "experiments" : [

  ]
}
}

### Operating system

macOS monterey 12.6.1

### Printer model

n/a

Manfredo1952, 2022-11-06:
Library not loaded: '/usr/local/opt/zstd/lib/libzstd.1.dylib  - this is what I can see - what happens if you copy that file from SS app contents fiolder (MacOS9 into that folder mentioned above ?

gopherspidey, 2022-11-23:
I fixed this by doing a "brew install zstd"