supermerill/SuperSlicer #3407: Unable to connect to printer

FlatEarthDave, 2022-11-06:
### Version
2.4.58.5

### Operating system type + version
Windows 11

### Behavior
I cannot connect to the printer, and this has been happening since upgrading to 2.4 from 2.3. I am able to upload to the printer through fluidd just fine, so there are no connection issues on my network. Also, my other install of SS 2.4.54.5 on Windows 11 connects just fine with the same settings. I installed it through Chocolatey on both systems. I just keep getting connection timed out with this one.  To be clear, it connected fine before I upgraded to 2.4 from 2.3

jhlchu, 2022-11-06:
What are your physical printer settings?
I'm uploading to fluidd from SS 2.5.59 no issue, so 2.5.59 worth a try. Also SS doesn't use fluidd user credentials, so if you have that it'll prevent upload from SS.

FlatEarthDave, 2022-11-06:
Turns out after several years, that RPi decided it didn't want the same IP address it has always been assigned. Sorry.

closed 2022-11-06 by FlatEarthDave