supermerill/SuperSlicer #3411: Ability to add multiple 'pauses' in the gcode viewer, without having to reslice

simpat1zq, 2022-11-07:
So I have a file that takes about 5 minutes to slice. And I need to do 5 color changes in there. However, when I add a 'pause' in there, the gcode viewer goes away, and I have to spend another 5 minutes to reslice so I can add my next pause. Is there any way to add all my pauses in there and then just do one slice at the end?

misiek271, 2022-11-07:
yep turrn off auto slicing then you can slice it manually add gcode and reslice it.

simpat1zq, 2022-11-07:
I had already turned that off. But as soon as I add a pause the gcode viewer's slider disappears until I hit the slice button again. Turning that off doesn't really help here.

Durahl, 2022-11-23:
Yea... I feel the pain. I get super anal about editing stuff with height range modifiers which was EZPZ because the view wouldn't reset thus allowing you to make all changes in one go but then something was changed and now it's a R.F.P.I.T.A.

Both the 2.4 and 2.5 versions now **_both_** having Game Breaking bugs when it comes to Multi Material Printing making one outright unusable for that task and the other a pain to deal with by manually fixing the G-Code once exported isn't helping either.

As much as I love the Only One Perimeter on Top / Bottom Layers feature it's hard to keep recommending SS over PS 😑