supermerill/SuperSlicer #3416: After Slicing G-Code looks Corrupted

Jackmnb314, 2022-11-09:
### What happened?

All the sudden my slicer started to output g code that looks like this. ( see attached picture) It started on only one Saved "Print Setting" Preset but now happens to each one. Also happens to every stl/3mf/obj file I try, even the presaved calibration cube. I have restarted my computer, cleared the cache, erased the program, and redownloaded. I am currently running the   2.5.59 Version. I do not have a bug report as it does not detect an issue. Its not just a visual glitch in the program, the gcode is actually outputting this way. ( Verified with cura, prusa slicer, and even tried printing the gcode.)

The odd thing is if I slice the file and click on the "Sliced Preview" tab, it shows the model sliced as I would expect.
I have no clue what is going on so any help would be fantastic.

Let me know if you need more info and I will be happy to provide.

Picture #1 (Bad Slicing)
[image]

Picture #2 (Correct view in "Sliced Preview" tab)
[image]

### Project file & How to reproduce

I have no clue how to help someone re-create my problem.

### Version

2.5.59

### Operating system

Windows 10

### Printer model

Ender 3v2 4.2.2 Board

neophyl, 2022-11-09:
Could you attach a saved project file that is showing the issue please as requested when opening a new issue.
Remember to zip the saved 3mf file or it wont be attached.

The reason a project is required is that it contains ALL the settings you are slicing with and so should in theory allow anyone to reproduce your problem and then possibly debug it.

jhlchu, 2022-11-09:
When you print the file, does it look like picture 1 or picture 2?
Try reslicing with all default profiles, Print, Filament, and Printer. Use any built in preset to rule out profile issues.
Did you make any changes to your graphics card or graphics driver (if visual issue).

Jackmnb314, 2022-11-10:
[Problem Project.zip](https://github.com/supermerill/SuperSlicer/files/9976078/Problem.Project.zip)

Here is my Project. When I print it looks like picture 1. Ive tried switching to all default profiles and it looks like my problem is with my saved "Printer Settings". Im still not sure the problem. I will try right now to remake the profile copying the exact settings and report back.

Jackmnb314, 2022-11-10:
Well, I believe I found it...
[image]
It looks like I somehow changed the g code precision to "0".  I changed it back to "3" like the other default profiles and it seems to work fine now. Thank you guys for helping me work this out, you are much appreciated!

jhlchu, 2022-11-10:
Close issue if it's solved.

closed 2022-11-10 by Jackmnb314