supermerill/SuperSlicer #3417: config bundle not imported completely

Jones1403, 2022-11-09:
### What happened?

I first exported a config bundle with physical printers on one PC running superslicer v2.4.58.5.
Next I imported that config bundle on another PC also running v2.4.58.5 of super slicer

for some unknown reason, not all settings are imported. Not even all printers are imported. 
The filament presets are imported, but none of the slicer presets nor the physical printers are imported.

I removed all files in %AppData%\SuperSlicer and retried to have the same result.

Superslicer however says it imported all configs. 
I attached the config bundle for debugging purposes

### Project file & How to reproduce

[SuperSlicer_config_bundle.zip](https://github.com/supermerill/SuperSlicer/files/9972663/SuperSlicer_config_bundle.zip)

### Version

2.4.58.5

### Operating system

windows 10 21H2

### Printer model

voron 2.4 350mm with afterburner

neophyl, 2022-11-09:
A config bundle only exports presets that are non standard. So if for example you had a stock PRUSA mk3 printer installed and some custom filament and print presets and then save a config bundle then when you import the bundle into a different install it only contains those custom ones.  You have to run the wizard to get anything that is installed through that normally.

Jones1403, 2022-11-10:
Thanks for your answer @neophyl 
I agree that only custom presets are exported, however everything that was configured in the original 2.4.58.5 installation was customized. In addition the export-import of the same presets worked in SuperSlicer 2.3.x.x (don't know which version it was)
I used the export bundle function to create a custom configuration for SuperSlicer that I can share with my colleagues.