supermerill/SuperSlicer #3420: Question:  Why is "Max Print Speed Locked"

inspectionsbybob, 2022-11-10:
### Version
2.4.58.5

### Operating system type + version
Win10

### Behavior
In trying to "tune" my system for best speed, acceleration, jerk and other things, I run into "max Speed" being LOCKED, or it won't let me edit that field.  What/where is the setting that is causing that to be locked?  I thought I looked in Filament, printer maximums, and other places.

_Is this a new feature request?_
No

#### Project File (.3MF) where problem occurs
Built-In Calibration Cube.

seansch, 2022-11-10:
I believe Max Print Speed is only used if you have Volumetric Speed enabled(set to non-zero).

AndrewEllis93, 2022-11-10:
Yes, and the max print speed actually does not take effect unless you have a print speed set to 0, in which case it will dynamically adjust speeds to try to maintain a more even volumetric flow rate. It does not simply cap the speeds like the volumetric speed does.

jhlchu, 2022-11-10:
Capping top speed with or without autospeed is found in `Filament Settings > Filament > Print speed override > Max Speed`.

AndrewEllis93, 2022-11-10:
> Capping top speed with or without autospeed is found in `Filament Settings > Filament > Print speed override > Max Speed`.

Merill changed the behavior from PS but it's sort of an undocumented thing. Or I think it was in a change log somewhere, but never made it to the tooltip. Even Merill forgot about it, lol: 
<https://github.com/AndrewEllis93/Ellis-SuperSlicer-Profiles/issues/10>

inspectionsbybob, 2022-11-11:
Closing as explained

closed 2022-11-11 by inspectionsbybob