supermerill/SuperSlicer #3421: libopenvdb.so.9.1: cannot open shared object

Big-Al81, 2022-11-11:
### What happened?

superslicer: error while loading shared libraries: libopenvdb.so.9.1: cannot open shared object file: No such file or directory

expected superslicer to launch

### Project file & How to reproduce

Fails to load when launching from desk top. 

From terminal type : superslicer
returns: superslicer: error while loading shared libraries: libopenvdb.so.9.1: cannot open shared object file: No such file or directory

All system files are up to date.  Error started after installing system updates earlier this week. 

investigating "openvdb 10.0.0-1" found installed on system. 

### Version

1:2.4.58.5-2

### Operating system

manjarolinux 22.0.0 Sikaris

### Printer model

cr10s5 heavily modified

Damien2s, 2022-11-13:
Exact same issue here.
Same version, same os
libopenvdb.so.9.1 is missing, only libopenvdb.so.10.0. Tried to create a link libopenvdb.so.9.1 to libopenvdb.so.10.0 and of course it doesn't work:
superslicer: symbol lookup error: superslicer: undefined symbol:_ZNK7openvdb4v9_12io19DelayedLoadMetadata7getMaskEm

exetico, 2022-11-13:
Same thing here. On Arch it should be part of the `openvdb` package - but here I'm also rocking 10.0.0-1...

Damien2s, 2022-11-16:
OK problem solved with version 1:2.4.58.5-3 on Manjaro

Big-Al81, 2022-11-17:
Thank you Guys.  I opted to wait for the version release which has fixed the the issue.

closed 2022-11-17 by Big-Al81