supermerill/SuperSlicer #3425: No perimeters on bridge areas and "Fill the voids with bridges" not triggering (due to small size?)

gkatev, 2022-11-12:
### What happened?

Hello, I have a model with some slots for screw/screw-heads -- the way I'm printing it there are some perimeters for these holes in the air. My classic approach of setting "No perimeters on bridge areas" to "Fill the voids with bridges" does not seem to be working with this model. Allegedly, it kind of looks like the feature does not trigger for small such features?

The model:
[image]

From under: 
[image]

Flying perimeters, without their voids filled
[image]

I'm saying that it might be due to the holes' size, because the feature still works for large models, e.g.:
[image]

Futhermore, If I set the perimeter width to something quite small, the features starts working for some of the holes. Intuitively, I imagine that with small holes, the overhang is only 1/2 lines and that does not trigger the "No perimeters on bridge areas"? Is it supposed to work like this? If yes, is there some way to specify the threshold for "No perimeters on bridge areas" to trigger, or to force applying it at certain areas (e.g. via modifiers)?

Thanks

### Project file & How to reproduce

[orangepi-mount.3mf.zip](https://github.com/supermerill/SuperSlicer/files/9994745/orangepi-mount.3mf.zip)

### Version

2.4.58.5

### Operating system

Arch Linux

### Printer model

Custom Prusa i2

jhlchu, 2022-11-13:
This took some time to investigate but I got this behavior to trigger on and off. The culprit in my case was `Print Settings > Infill > Advanced > Anchor solid infill by X mm` I set it to `0` to fix but you probably need a value higher so that it touches the side. Slowly drop the value until you get it to trigger again. 
I tried it on various hole sizes with different overhangs, it's consistent.

jhlchu, 2022-11-14:
I forgot to specify it's the `Bridge` value for `Anchor solid infill`.
[image]
[image]

jhlchu, 2022-11-14:
It's because `~0.6mm External Width * 200% > Internal Circumference of Hole`. It can't create the bridge with your specified parameters so it just skips it.

gkatev, 2022-11-15:
Ah, thanks, indeed lowering this value fixing this problem! Though, at 1% the 5th hole (the nut trap, at the side of the model) still does work, and at 0 SS crashes...

But I'm afraid I still haven't fully grasped the reason this occurs! When you say "It can't create the bridge", does "bridge" refer to the void-fill (light blue, marked as "Bridge infill")? And I'm not sure I follow the math either; the hole diameter is ~3.2mm, so its circumference is ~10mm > 0.6 * 2.

In the middle of all this, I'm getting crashes when setting some perimeter width values (#3432); there are some perimeter widths for which the problem is fixed, e.g. 0.7mm, but for only 2 of the holes (!), but then with 0.5mm perimeter width it again does not work as expected.

jhlchu, 2022-11-15:
Try with 2.5.59 nightly, it's been less crash-ey for me than 2.4.58 stable.
https://github.com/supermerill/SuperSlicer/actions/runs/3047588416
https://github.com/supermerill/SuperSlicer/actions/runs/3047588404

gkatev, 2022-11-15:
Thanks, indeed the crashes went away. With the value set to 0 the problem is fixed for the 4 holes (not for the fifth though).

-> Do we consider this phenomenon a bug?