supermerill/SuperSlicer #3427: Auto arrange pushes parts off plater when space exists

OldCurmudgeon3DP, 2022-11-12:
### What happened?

Unclear if this is really a "feature"(bug) or a profile issue.

When using auto arrange button a part is frequently pushed off the surface even though there is plenty of room for it if I manually arrange the pieces.
[image]

### Project file & How to reproduce

[C_Jaw.zip](https://github.com/supermerill/SuperSlicer/files/9995197/C_Jaw.zip)

### Version

2.5.59.0 (nightly 617)

### Operating system

Win 10

jhlchu, 2022-11-13:
What do you have for `Printer Settings > General > Bed Shape`. In your project you have it set as `Custom`. I changed it to `Rectangular` and it fixed the arranging issue, just input the size for your bed.

OldCurmudgeon3DP, 2023-01-12:
It's a custom because I have a collision potential in the back left corner. The bed image has a notch cut out in that area. Wouldn't think that would have an effect though.