supermerill/SuperSlicer #3428: I can’t install SuperSlicer on my Mac

bjoernbarth, 2022-11-13:
### What happened?

If I start the Slicer my Mac shows me the following “SuperSlicer cannot be opened because the developer cannot be verified.”

How can I solve the problem?

SuperSlicer version 2.4
MacOS 13.0

### Project file & How to reproduce

Not needed 

### Version

the latest 

### Operating system

macOS 13.0

### Printer model

not needed

Naninani, 2022-11-14:
Looks like the packages are not signed and notarized by the Mac Notarization service. Without that no one will be able to run the the software.

mjoconr, 2022-11-14:
You can override this by going to System Preferences -> Security & Privacy then press open anyway.

plundrigan, 2022-11-18:
I could never get the latest version to open. even using the cmd xattr -cr /Applications/SuperSlicer.app  Oddly, it acts like it wants to launch and then closes. I did install the archane version and was able to get that to work.

bjoernbarth, 2022-11-18:
> You can override this by going to System Preferences -> Security & Privacy then press open anyway.

I can't find the settings.

plundrigan, 2022-11-18:
> > You can override this by going to System Preferences -> Security & Privacy then press open anyway.
> 
> I can't find the settings.

This will only show when the app is first launched. It is sometimes delayed. It is will show directly under the allowed apps downloaded from radio buttons.

I think this command xattr -cr /Applications/SuperSlicer.app  assuming you placed the App file in Applications is designed to help bypass the check.  I am having a hard time finding exactly what the -cr attributes are for.

errata79, 2022-11-21:
Hello friends, I have an intel mac and I can not install the current version 2.4.58.5. I've tried from command and I can't.
Version 2.5.59.0 is fine.
can someone help??
I leave the error that launches version 2.4.58.5.

Translated Report (Full Report Below)
-------------------------------------

Process:               SuperSlicer [1926]
Path:                  /Applications/SuperSlicer.app/Contents/MacOS/SuperSlicer
Identifier:            com.merill.slic3r/
Version:               SuperSlicer SuperSlicer_2.4.58.5_2022-09-23 (SuperSlicer_2.4.58.5_2022-09-23)
Code Type:             X86-64 (Native)
Parent Process:        launchd [1]
User ID:               501

Date/Time:             2022-11-21 02:47:47.5724 +0100
OS Version:            macOS 12.6 (21G115)
Report Version:        12
Bridge OS Version:     3.0 (14Y910)
Anonymous UUID:        48FA12A0-71CA-88B8-9890-376A36985583

Sleep/Wake UUID:       811D4B42-0E3C-45F1-A9D5-7293F8B8ABC1

Time Awake Since Boot: 6400 seconds
Time Since Wake:       1047 seconds

System Integrity Protection: enabled

Crashed Thread:        0

Exception Type:        EXC_CRASH (SIGABRT)
Exception Codes:       0x0000000000000000, 0x0000000000000000
Exception Note:        EXC_CORPSE_NOTIFY

Termination Reason:    Namespace DYLD, Code 1 Library missing
Library not loaded: '/usr/local/opt/zstd/lib/libzstd.1.dylib'
Referenced from: '/Applications/SuperSlicer.app/Contents/MacOS/SuperSlicer'
Reason: tried: '/usr/local/opt/zstd/lib/libzstd.1.dylib' (no such file), '' (no such file), '' (no such file)
(terminated at launch; ignore backtrace)

Thread 0 Crashed:
0   dyld                          	       0x117ed898e __abort_with_payload + 10
1   dyld                          	       0x117eeea72 abort_with_payload_wrapper_internal + 80
2   dyld                          	       0x117eeeaa4 abort_with_payload + 9
3   dyld                          	       0x117e9c699 dyld4::halt(char const*) + 615
4   dyld                          	       0x117e985b5 dyld4::prepare(dyld4::APIs&, dyld3::MachOAnalyzer const*) + 3995
5   dyld                          	       0x117e974e4 start + 388

Thread 0 crashed with X86 Thread State (64-bit):
  rax: 0x0000000002000209  rbx: 0x0000000000000000  rcx: 0x00007ff7b477c628  rdx: 0x00007ff7b477ca90
  rdi: 0x0000000000000006  rsi: 0x0000000000000001  rbp: 0x00007ff7b477c670  rsp: 0x00007ff7b477c628
   r8: 0x00007ff7b477c690   r9: 0x0000000000000000  r10: 0x0000000000000075  r11: 0x0000000000000246
  r12: 0x0000000000000075  r13: 0x00007ff7b477ca90  r14: 0x0000000000000001  r15: 0x0000000000000006
  rip: 0x0000000117ed898e  rfl: 0x0000000000000246  cr2: 0x0000000117e9eda4
  
Logical CPU:     0
Error Code:      0x02000209 
Trap Number:     133

Binary Images:
       0x117e92000 -        0x117efdfff dyld (*) <71febccd-d9dc-3599-9971-2b3407c588a8> /usr/lib/dyld

External Modification Summary:
  Calls made by other processes targeting this process:
    task_for_pid: 0
    thread_create: 0
    thread_set_state: 0
  Calls made by this process:
    task_for_pid: 0
[... 103 lines cut ...]
        "SIRI_DICTATION_ASSETS" : "620aec83b02b354d3afd2f50"
      },
      "deploymentId" : 240000145
    }
  ],
  "experiments" : [

  ]
}
}

Model: MacBookPro13,3, BootROM 451.140.1.0.0, 4 processors, Quad-Core Intel Core i7, 2,7 GHz, 16 GB, SMC 2.38f12
Graphics: Intel HD Graphics 530, Intel HD Graphics 530, Built-In
Graphics: Radeon Pro 455, AMD Radeon Pro 455, PCIe, 2 GB
Display: Color LCD, 2880 x 1800 Retina, Main, MirrorOff, Online
Memory Module: BANK 0/DIMM0, 8 GB, LPDDR3, 2133 MHz, 0x802C, 0x4D5435324C31473332443450472D30393320
Memory Module: BANK 1/DIMM0, 8 GB, LPDDR3, 2133 MHz, 0x802C, 0x4D5435324C31473332443450472D30393320
AirPort: spairport_wireless_card_type_wifi (0x14E4, 0x15A), Broadcom BCM43xx 1.0 (7.77.111.1 AirPortDriverBrcmNIC-1710.4)
AirPort: 
Bluetooth: Version (null), 0 services, 0 devices, 0 incoming serial ports
Network Service: Wi-Fi, AirPort, en0
USB Device: USB30Bus
USB Device: Apple T1 Controller
Thunderbolt Bus: MacBook Pro, Apple Inc., 41.5
Thunderbolt Bus: MacBook Pro, Apple Inc., 41.5

unixb0y, 2022-12-14:
In my case, the zstd issue came from an alternative Homebrew install location that I use.
Nonetheless, that's still a bug in SuperSlicer. It should look for zstd in all paths and not onl in /usr/local/opt.

Please fix! ;)

errata79, 2022-12-14:
> In my case, the zstd issue came from an alternative Homebrew install location that I use. Nonetheless, that's still a bug in SuperSlicer. It should look for zstd in all paths and not onl in /usr/local/opt.
> 
> Please fix! ;)

Have you already solved it?

unixb0y, 2022-12-14:
> 

Kinda...

Find your zstd and link it.
`
mkdir /usr/local/opt
ln -s $(brew --prefix zstd) /usr/local/opt/zstd
`

unixb0y, 2022-12-14:
> > In my case, the zstd issue came from an alternative Homebrew install location that I use. Nonetheless, that's still a bug in SuperSlicer. It should look for zstd in all paths and not onl in /usr/local/opt.
> > Please fix! ;)
> 
> Have you already solved it?

Kinda...

Find your zstd and link it.
`
mkdir /usr/local/opt
ln -s $(brew --prefix zstd) /usr/local/opt/zstd
`

errata79, 2022-12-14:
> 

woo.. I'm not that expert. Sounds Chinese to me.