supermerill/SuperSlicer #3434: Feature request - Maintain extruder order on layer change

DocRuzzy, 2022-11-16:
I am using two objects overlayed so that I can print a UV-curable material with one head and then retrace the same path with a UV light source at a slower rate.

This works fine for a single layer, but the slicer will keep the currently active head on a layer change, which causes the wrong extruder order on multiple layer prints:

layer 1.
head 1 prints. (curable gel)
[single uv circle.zip](https://github.com/supermerill/SuperSlicer/files/10024348/single.uv.circle.zip)

tool change
head 2 prints (UV)
layer change
head 2 prints (UV)
head 1 prints (curable gel).

I am requesting a feature that includes a checkmark setting to maintain head or object order (but not 'complete individual objects') would allow me to perform multilayer UV cure prints.

Even greater would be extending this feature to define an extruder order. 

#### Project File (.3MF) where problem occurs
[single uv circle.zip](https://github.com/supermerill/SuperSlicer/files/10024351/single.uv.circle.zip)

lordramzes, 2024-04-06:
I'm raising the issue. This option would be ideal for multi-color printing with a multi-head printer. This would allow you to ALWAYS maintain the same even flow after changing the tool. Unfortunately, when printing according to the scheme, for example Layer-1: T1-T2-T3, Layer-2: T3-T1-T2, Layer-3: T2-T1-T3 etc., the next layer using the previous tool always has a slightly different flow , which is quite visible on the printout, like a staircase effect.

DocRuzzy, 2024-04-06:
I think we need to take this over to the Prusaslicer repository. I thought I did but can't find it now.

lordramzes, 2024-04-07:
I think that in general it will be difficult to interest someone in this problem, not to mention implementing the correction :) From conversations with "local" people it turns out that I was the only one who noticed the problem with uneven flow in multi-head printing using this procedure, that the last head on previous layer is the first head on the next layer. In my opinion, the modification we are talking about probably goes quite deep into the program's mechanics, but I think that Supermerill would be able to add an appropriate option on its own. On the other hand, Prusa should actually be interested in this solution since it released the multi-head XL version. I tried to solve this problem on my own (you can cheat the slicer's algorithm a bit), but it is not as optimal a solution as implementing it correctly from the beginning.