supermerill/SuperSlicer #3437: 2 months without updates!?

user897943, 2022-11-17:
### What happened?

I thought this was based on PrusaSlicer, so where is the lightning infill for example? Arachne still "pre-release"... Should I move back to PrusaSlicer, because this project seems to be dead? Would be great to know if it's deprecated/no longer maintained.

### Project file & How to reproduce

See above

Creo2005, 2022-11-17:
:(

doctor-bob, 2022-11-20:
i get the same feeling about this...almost like theres something wrong since even before the latest release.

TravisWilder, 2022-11-20:
I asked merill and he replied on 24.10.2022
"I had to focus on something else this month. Hopefully, i can focus back on the slicer now."

So it seems he is still busy

hvgithub2, 2022-11-20:
Do you use PrusaSlicer with your Bambu Labs printer ? Are there some advantages to that slicer ?

doctor-bob, 2022-11-21:
i feel bad for not being able to help(working on it), THANKS to everybody involved in development and troubleshooting!

BeunHaas34, 2022-11-23:
Keep in mind: This a open source project. Its not a job

user897943, 2022-11-23:
> Keep in mind: This a open source project. Its not a job

I gotta nip this right in the bud. We ALL appreciate someone's work and contribution but no, **it's definitely a job**. If someone's publishing a project on GH with the description below (1), then they are basically assuming responsibility for maintaining. In software terms 2 months is like 3 years without updates and when the project also states that it pulls upstream changes from PrusaSlicer there is no good excuse for the delay. Having personal issues to deal with is an acceptable excuse not a good one.

The solution is to either get back to supporting updates, or release the project for an active maintainer to take care of it, or archive the project. But it is UNFAIR to users to keep the project open indefinitely while they hope updates are coming but nothing is certain.

1: 
```
Can I help?
Sure! You can do the following to find things that are available to help with:

Add an issue to the GitHub tracker if it isn't already present.
Before sending patches and pull requests contact me (preferably through opening a GitHub issue or commenting on an existing, related, issue) to discuss your proposed changes. This way we can ensure that nobody wastes their time and no conflicts arise in development.
```

hiaselhans, 2022-11-25:
@user897943 you'd better start your own opensource project, then come back here in five years and apologize.

People tend to have a life. If you want them to spend their time on your problems, pay them.

user897943, 2022-11-25:
That's exactly what I did yesterday. I gave Bambu Lab around £1500 of my
money for a fully supported proprietary product that will be fully
supported. And you enjoy your unsupported slicer and your nasty attitude
toward people who took their time to trust, use, and contribute to this
project only to have their pull requests put on hold by ONE person because
you know, all these pull requests weren't made by people who have a life
/sarcasm
Like I said, I'd the project owner has no time THEN WALK AWAY. If it's open
source he doesn't own it, the community does. You are part of the problem.
I'm unsubscribing to enjoy my new  closed source fully supported
acquisition without bad attitude from people like you.

BHoggs, 2022-11-26:
@user897943 You seem to be confusing commercial open source products with a personal project. merill has stated before that this is his personal project that others are welcome to use. It's not "owned by the community", again, it's merill's pet project. "Open source" does not in any way imply the project belongs to the community.

It does however mean the community are free to fork it, and turn it into a community run project.

Basically if you don't like it, go use something else or make your own. (which you have, but you have no right to complain about someone's personal fork maintained in their own spare time.)

Arakon, 2022-11-26:
Wow, that's some entitlement.
I bet you also scream at minimum wage cashiers that you own their asses and will get them fired.

This is a free project, done in a private person's private time. A person who has a job, a person who has family, a person that has other priorities than making an entitled clown on the internet happy cause there's been a whopping 60 days without an update. DURING A HEALTH CRISIS.

Be the maintainer you want to see while taking your toys and playing elsewhere, but don't bitch at people who have already spent hundreds of hours working on this for free.

user897943, 2022-11-26:
> Wow, that's some entitlement. I bet you also scream at minimum wage cashiers that you own their asses and will get them fired.
>
> This is a free project, done in a private person's private time. A [...]

Hahahahahaha
I love the personal attack, those always work so well.
Health crisis 😭 lololol you can't milk this cow forever.
I'm a gov worker myself and almost died from Covid during the first wave (wouldn't surprise me if you're the type of person who would respond something evil like "should have", you just come across like that).
As for your guess, that's the wrong guess. I treat them with the utmost respect, with a smile, and I'm always the first to ask how their day is going. I joke with them and make them laugh. I'm a socialist. I believe in being kind. I don't think it's right to keep a project running where people depend on it to make their living but even with sponsorships and others creating pulling requests the project is totally ignored for TWO MONTHS. So shut up, you're the clown.

closed 2022-11-26 by user897943

jtrmal, 2022-11-26:
And we, as a community are very happy to lose you, @user897943 
I hope life will work out well for you either way and good luck.

shaneshort, 2022-12-01:
have a gander at this guys github "contributions".. he's already started bitching out the bambu guys and all he does is raise issues on projects.

Eldenroot, 2022-12-03:
Thats why we should have only PrusaSlicer... all features in a one package. Now we have two slicers based on the same core - PrusaSlicer and SuperSlicer. I love the expert settings in SuperSlicer, but many features are different from PrusaSlicer (at least every version differs more and more...).

Maybe would be better to contribute all these changes into PrusaSlicer and maintain only one package. Also there could be a new mode - Expert+ 

It would contain all extra settings from SuperSlicer :)

Yeah, really nice dream, maybe sometime, someday :) who knows ;)

Keep good work!

user897943, 2022-12-10:
How you all doing with your buggy and forgotten-about slicer, ass-kissers? I'm enjoying my X1-Carbon and proprietary fully-supported slicer it came with 😭 lolol

user897943, 2022-12-10:
> have a gander at this guys github "contributions".. he's already started bitching out the bambu guys and all he does is raise issues on projects.

Yesssss go take a look at my page and don't forget to leave a comment, like and subscribe 🤣🤣🤣🤣 

You don't understand how being a customer works, you're retarded. Something that doesn't work needs to be reported to be fixed, moron 🤣🤣🤣🤣

BHoggs, 2022-12-10:
@user897943 just move on. I've reported you for your highly disrespectful conduct.

user897943, 2022-12-10:
> @user897943 just move on. I've reported you for your highly disrespectful conduct.

Okay and? 🤣🤣🤣🤣 I'll report myself to help you.

rufinus, 2022-12-10:
@user897943 users like you are the reason people stop giving great stuff to the community. your audacity is beyond words. From now on i name you kevin, thats what you are. Where are you a customer? on SuperSlicer? dont think so... on esphome? neither. at Bambu? yeah maybe. then go play on their support forum here, allow me to show you the way. https://forum.bambulab.com/

@supermerill i know it isn't much but you have my personal thanks for the work you do for us. If you dont have the time thats ok. I know that a forked project is hard to keep up with the origin.

user897943, 2022-12-10:
> @user897943 users like you are the reason people stop giving great stuff to the community. your audacity is beyond words. From now on i name you kevin, thats what you are. Where are you a customer? [...]

Nope. This has been abandoned for 2 whole months. Maybe 3 now. With zero updates. It's so dismissive of the developer. THEN users like me who put their faith in this app come to ask for an update and get treated like crap by ass-lickers like you. Have your abandoned app then! Enjoy it!

rufinus, 2022-12-10:
@user897943 kevin, i neither have the time nor the crayons to explain you how opensource works. just leave us alone please.

user897943, 2022-12-10:
> @user897943 kevin, i neither have the time nor the crayons to explain you how opensource works. just leave us alone please.

Don't have the time but typed that useless comment... Pathetic

Misterke, 2022-12-21:
I also hate that SuperSlicer has come to a standstill, but honestly the real problem I feel is that PrusaSlicer is not pulling the improvements added by SuperSlicer upstream.  So, instead of complaining here on supermerill not updating this anymore, perhaps we should all be bugging the PrusaSlicer team to provide the features we all love in SuperSlicer.

user897943, 2022-12-23:
How y'all ass lickers doing with your outdated slicer? Loving Bambu Studio getting meaningful updates while I laugh at my "audacity" with y'all! By the way, each downvote = 1 laugh

pemerick07, 2023-01-04:
@user897943 You're crazy man. I was wondering what was going on with SS updates too because I love the slicer and I've actually donated money to Merill, but yet expect less from him than you. I also use Bambu Studio for my X1C, but yet it doesn't have enough control over settings to get air-tight parts, so I need to use SS with my Voron for that.

user897943, 2023-01-04:
Totally normal person here. Tho I find it fun to mess with ass-lickers who come pontificate on someone's bug report.

aidbish, 2023-02-10:
Hes' the same dickhead over at HA project. Seems an entitled muppet

user897943, 2023-02-12:
> Hes' the same dickhead over at HA project. Seems an entitled muppet

I just loooooove how sad losers like you go through my post history to validate your theories. You are partially right. I am extremely entitled. I expect quality and timely updates for any software I use or for the maintainer to just grow a pair and transfer ownership to someone who actually has the time to maintain it, or to just archive the damn thing if they can't be bothered. You are the muppet for not demanding the bare minimum from projects on this site. So much junk here from people who just can't be bothered, just causing hours and hours of frustration to others. Enjoy your out of date and buggy slicer, loser. And please bookmark my page so you can foam at the mouth when I post something new LMAO