supermerill/SuperSlicer #3441: ABS shrinks in the Z direction too: we need a shrinkage parameter for that

firsttest12, 2022-11-21:
### Version
_Version of SuperSlicer used goes here (help->about)_

2.4

### Operating system type + version
_What OS are you using, and state any version #s_

OSX

### Behavior
* _Describe the problem_

Filaments shrink in the Z direction too. (This is very noticeable if you print ABS and then take it out of the hot build chamber.)
We need a compensation value for the Z axis. In my experience the value is less than for the XY, but it is still an important value. Without it my parts will not fit together and will have to make compensations in CAD. 

Note: Simplify3d has this feature (scaling in X, Y, Z to allow for shrinkage compensation).

* _Steps needed to reproduce the problem_
* _Expected Results_
* _Actual Results_
  * _Screenshots if useful_

_Is this a new feature request?_
Related guides for writing feature requests: http://meta.stackexchange.com/a/259196 http://nickohrn.com/2013/09/write-great-feature-request-bug-report/

#### Project File (.3MF) where problem occurs
_*Upload a Project File* (.3MF) `Plater -> Save Project as...`_
_You have to zip it (or rename it to .zip) to be able to drag& drop it here_

Garryb123, 2022-12-10:
I think this is an important request because models with complicated internal geometry cannot be adjusted in CAD without superhuman effort but, can the present model resizing in XYZ help you?

firsttest12, 2022-12-11:
Hi, I thought this was a feature request, but in fact it's a bug report.

You are quite right that the "object manipulation" pane for an individual model does allow resizing of the x, y, z dimensions independently, which is exactly what I was requesting.

HOWEVER the "object manipulation" pane for a Group of objects (that is, if you have placed an model, then inserted other models in association with it behaves differently:

Changing the default "100%" scaling on one of the dimensions will result in:

a) the model scaling in all three dimensions according to the number that was entered. It is not possible to "unlock" the padlock that otherwise would unlink the scaling of the three dimensions;

b) despite the group of models having scaled, in each of the dimension boxes it will remain "100%". So, if you just entered "150%" all models in all dimensions will scale to 150% BUT in the scaling boxes it will still say 100%;

c) in addition, examining the "object manipulation" panes for each of the models in the group, I find that not all of the scaling boxes have changed from their initial 100% (but everything has scaled nonetheless);

So, a bug report. 

Thanks

neophyl, 2022-12-11:
What method have you been using to compensate for shrinkage ?  As you haven't attached a 3mf project file as requested when opening an issue the developer wont know what you are working with.  No pics, or any of the requested info on opening a new issue in fact.

Have you been using x/y compensation in the print profile or have you tried the filament shrinkage setting in the filament profile ?  No one knows, except you and you aren't communicating that info.

btw the right hand panel scaling issues are inherited from PS and from what I understand will not be easy to fix.  Not without the prusa team doing it anyway.  Feel free to go add your support to the open issue(s) in the PS github.
For example https://github.com/prusa3d/PrusaSlicer/issues/8009
or https://github.com/prusa3d/PrusaSlicer/issues/6122
or https://github.com/prusa3d/PrusaSlicer/issues/4807
or https://github.com/prusa3d/PrusaSlicer/issues/3789

Hopefully you see my point about this issue.  In all this time this has not been tackled by prusa and they have a whole team of devs.  SuSi has a single main developer and a few others who contribute.

firsttest12, 2022-12-11:
totally understand

I'm new to Superslicer - and skipped Prusa slicer - so didn't think to consider who does what.

Sorry.