supermerill/SuperSlicer #3442: "Reverse on odd" has no effect on the print at all

VanessaE, 2022-11-22:
### What happened?

Title says it all :smile: 

I'm trying to print a small part with horizontal holes, essentially using it as a calibration piece to reconfigure my 0.1 mm layers profile so that I can print more of them at this layer height, and am getting really messy corners and overhangs (my prints look good with 0.2 mm layer heights though).

One of the things I had hoped to employ to clean the overhangs up (and maybe the corners too) was to enable "reverse on odd", but no matter what "reverse threshold" I set, it just refuses to actually do anything to the print direction.

I tried setting the threshold as low as 0.01 and "bridge speed and fan" to a similarly tiny value of 0.01%, and while the latter does cause the "gcode preview" display to mark even the slightest overhangs as, well, overhanging, the print direction still doesn't change.

### Project file & How to reproduce

Project file:
[horiz-hole.3mf.zip](https://github.com/supermerill/SuperSlicer/files/10063589/horiz-hole.3mf.zip)

Slice the part, scroll the layer position to 6.5 mm, and compare the print direction between adjacent layers -- every layer goes counter-clockwise no matter what.

Note that the steepest overhang in this model is about 45°, as it uses a variation on the old "teardrop holes" shape (the real model uses them, but I can't publish it yet, so I made this test object to focus on just this one issue).  Maybe 45° isn't steep enough to trigger this feature?

Also note: this project contains untested changes, the result of a lot of trial and error, but only the "reverse on odd" feature is at issue here.

### Version

2.4.58.5, using the official "late 2.4 edition" release Appimage 

### Operating system

Debian "Bullseye" 11.5.0

### Printer model

Custom-built Hypercube

VanessaE, 2022-11-24:
Okay, apparently this feature requires that "extra perimeters / on overhangs" be enabled, BUT even with that feature enabled, the print direction does not change.

My best guess is that even on a perfectly round hole, if it's small enough, there won't be any reason for the slicer to add more perimeters (because nothing internal is exposed).

Can we just have this turned into a force setting that relies only on its own threshold similar to the threshold for bridge speed/fan, regardless if extra perimeters are to be added?

VanessaE, 2022-11-24:
Edit to my last comment: 

Can we just have this turned into a force-enable setting that relies only on its own threshold (the one that already exists), and flips the direction on odd layers regardless if extra perimeters are to be added?

nathanshearer, 2023-07-06:
I am unable to reverse the printing direction on odd layers, with any setting. It seems this feature doesn't work at all.

RealDeuce, 2024-01-06:
Reverse on odd is for overhangs (ie: in the overhangs section)... to get it to apply to all perimeters, you need to set the "Reverse threshold" (overhangs_reverse_threshold) to zero.  Also, it doesn't work with Arachne, you need to use the Classic perimeter generator.

closed 2024-01-16 by VanessaE