supermerill/SuperSlicer #3444: add M118 to the available methods for "print remaining time"

Paradisebaker, 2022-11-22:
**Is your feature request related to a problem? Please describe.**
not really a problem. I have Octoprint and a TFT28 connected to a MKS GenL (Artillery X1).
M117 is not pushing the remaining time (and layers) from Octoprint via the board to the TFT.
Layer stats work via M118, as apparently the board acks the command and thus the TFT knows about it.
But layers are easy, just add the "M118 A1 P0 action:notification Layer Left [layer_num]/[total_layer_count]" to the layer change Gcode.
However, print remaining time is put automatically into the Gcode.
I can do a find replace in notepad++ to change
M117 Time Left 0h0m53s  -->> M118 A1 P0 action:notification Time Left 0h0m53s
but that is cumbersome.
same with having to edit all my print setting profiles for post-processing.

**Describe the solution you'd like**
Add the M118 syntax to the available methods in Printer settings > General > Firmware > "print remaining times"
[image]

**Describe how it would work**
I dont know, maybe it is as simple as adding another menu item with different syntax output for the gcode generation

**Describe alternatives you've considered**
none really worked for me, so far the only one working is notepad++ find-replace

**Additional context**
There seems to be the same request (I think even already implemented) for Cura, as in this github post:
https://github.com/Ultimaker/Cura/issues/10914

Paradisebaker, 2022-11-22:
[DisplayProgressOnLCD.zip](https://github.com/supermerill/SuperSlicer/files/10066687/DisplayProgressOnLCD.zip)

This is a python script that the OP in the Cura thread did implement.
Maybe it can be used as template for a Superslicer implementation.
Please note that he used P2 as default, this should either be an option that can be chosen or P0 should be used as default

neophyl, 2022-11-22:
Later super slicer versions like PS have a built in find replace that you can use.  You could just use that to replace the M117 lines with whatever you want.  No need to use a python script.

Paradisebaker, 2022-11-22:
Just for the sake of a workaround, I just tried the find-replace routine for my M117 to M118 replacement. It just did not work.
When googling for it, I found that someone else reported this in this forum (#2644).
Kind of funny, as you, neophyl, replied to that as well, stating that the M117 insertion in the printer settings is a post-process routine, which cannot be altered by the find-replace from the print settings.....

neophyl, 2022-11-22:
Just goes to show that memory is fallible.  I thought it had been sorted.

Paradisebaker, 2022-11-22:
> Just goes to show that memory is fallible. I thought it had been sorted.

oh, how I know about failing memory.... no offense!

Anyhow, to me that other thread shows that I am not the only one liking M118....

Paradisebaker, 2022-12-26:
can I have a wish for christmas?
please add M118.....

maniac75, 2023-08-05:
Maybe if someone can create a post procession script, that replace all the M117 Time Left ?h?m?s to M118 A1 P0 action:notification Time Left ?h?m?s it would work and we dont have to wait to changes in the app itself with a new release..
I tried to do it with G-code substitution but didnt know how to make it works and i dont know how to program in python, just found a post processing script that replace the m73 to m117 in https://github.com/ichekanov/PrusaSlicer_time_converter
If someone can create the script or figure out how to use gcode substitution, please share it, thanks

maniac75, 2023-08-06:
I modified the script, and it worked, with some glitches but it's ok
maybe someone with more python knowledge can improve it or make one to replace m117 to m118
I'll share the script i modified, thanks to @ichekanov
[m73 to m118 gumana.py.zip](https://github.com/supermerill/SuperSlicer/files/12269650/m73.to.m118.gumana.py.zip)