supermerill/SuperSlicer #3449: Pressure Equalizer from Prusa Slicer

meltiseugen, 2022-11-23:
I would love to have the newly added feature in Prusa Slicer: Pressure Equalizer

kursatu, 2022-11-29:
I did not know that there is a such a feature. I have been using my own post-processor that adjusts the 
E acceleration depending on the perimeter type.
It works pretty well. I am using dragon SF/HF hotends with 30-45mm cube/sec volumetric speeds.
It helps a lot with especially when the volumetric print speed, hence the pressure, is close to the maximum.
I've never found the linear advance/pressure advance of Marlin usable.
I though that it should be part of the slicer and this certainly improves a bit over my post-processor that adjusts E acceleration.

meltiseugen, 2022-11-29:
> I did not know that there is a such a feature. I have been using my own post-processor that adjusts the E acceleration depending on the perimeter type. It works pretty well. I am using dragon SF/HF [...]

This feature in PrusaSlicer is different: It adds gradients between changes in print head speed, and thus should result in smoother pressure changes. This would help a lot if you have issues at the start of a line with over or under extrusion.

Official documentation: https://help.prusa3d.com/article/pressure-equalizer-_331504