supermerill/SuperSlicer #3451: Modifiers work wrong for fuzzy skin

wolfgangmauer, 2022-11-24:
### What happened?

If i add a modifier to my part an disable Fuzzy Skin it seems not to work right. 
[image]
[image]

### Project file & How to reproduce

Add a modifier to a part with fuzzy skin enabled and disable fuzzy skon on the modifier.

### Version

2.4.58.5

### Operating system

Windows10

### Printer model

Voron 2.4 R2

robustini, 2022-11-24:
Duplicate https://github.com/supermerill/SuperSlicer/issues/1092

neophyl, 2022-11-24:
Its how modifiers work.  For most of the settings it in effect splits them into separate volumes to process, this means you get extra perimeters etc around them.  Its not just fuzzy skin.