supermerill/SuperSlicer #3455: Conical Slicer Implementation

SkyShadex, 2022-11-26:
**Is your feature request related to a problem? Please describe.**
Not really.

**Describe the solution you'd like**
There exists this [Conical Slicing work around from @RotBotSlicer and CNC Kitchen](https://github.com/RotBotSlicer/Transform) that I would love to implement into SS to avoid having two programs to manipulate the same file with.

I know it was designed to get around eliminating support material for overhangs. But I'm interesting in it's applications for better structural integrity in regards to layer orientation.

For instance, I'm designing a rather geometrically simple load bearing part. Because of the load, print orientation is rather important. Currently I have to orient the part for a desired angle, but that creates alot of support material.

This workaround does a great job of eliminating wasted support material and giving me better angles. But it involves exporting SS > 1st Python Script > SS > 2nd Python Script.

**Describe how it would work**
[This is how it currently works](https://www.cnckitchen.com/blog/guide-how-to-use-conical-slicing)

I would like it to happen entirely inside of SS, nothing against python, but I spent enough time learning linux as is.

- Orienting the model 
- transforming model with the script
- slicing the transformed model in SS
- gcode transformation via script

What I imagine the steps to accomplish this would be:

1. creating a macro of the steps involved in orienting the part and it's origin for the 1st script to transform
2. slicing angle defined in SS gets passed to script (I believe it's called cone_angle in the script)
3. running 1st python script inside/alongside of SS? (unsure how well python and C play together)
4. slicing the transformed model (nothing new here other than making sure the origin work done in step 1 is the same
5. running that gcode through the 2nd python script... Inside/alongside SS
6. ability to export that transformed gcode (as it does currently)

I suppose steps 3-6 could be a function of pressing export, from the end user's perspective.

Settings I'd like to see within the UI:
```
Print Settings > Slicing > "Conical Slicing" (or if you'd like to expand on slicing modes under Other)
Conical Slicing: enable, disable
slicing angle: x degrees. (maximum angle used here should not exceed clearances on your print nozzle and surrounding points on the print head. ex: fan shrouds, sensors, etc..)
slicing direction: inward, outward (k.i.s.s, the idea of full-featured variable conical slicing isn't my goal, just implementing this script)

```

**Describe alternatives you've considered**
doing it the current long way.

**Additional context**
Not sure if this sits within the realm of the script or related to how the part is oriented in regards to the origin.

In the case of my ender bedslinger, I have much better clearance angles on my Y axis than I do my X axis (sensors, shrouds. I imagine, the further from the origin the part is in any direction, the less "conical" this script would look and the more "non-planar" it would appear. So, slicing the part "inward", with a large y offset, and my desired slicing angle, should give me the ability to print with angles largely in the YZ plane (there would be an angle in the XZ plane, but they would be minimal depending on the offset). Akin to how belt printers like the CR-30 operate! (Except we're actively using the Z axis to replace the physical differences of belt printers.)

In this way, the part would be sliced at the desired angle, in an orientation that's suited for your best clearance angle

Variables to watch for:
Max YZ angle
Max XZ angle

```
Printer Settings > Extruder 
Define max X and Y clearance angles.

IF (conical slicing: enable)
THEN modify auto-arrange
GET user-defined x & y clearance angles
GET user-defined slice angle
GET user-defined slice orientation ? (maybe 
CALCULATE Offsets the origin appropriately 

```

jhlchu, 2022-11-26:
SS has field for post-processing script in *.py that automatically runs on export.
Or you can run SS (https://manual.slic3r.org/advanced/command-line) and *.py from terminal using a script.

SkyShadex, 2022-11-26:
Ah, that does cut down some steps from the current work around! I can certainly play around with that!

The issue is that it still requires you to export from SS twice. I suppose that can be mitigated by using the command line in a script. 

Except I use SS in Windows 10, the linux thing was me converting a phone into a klipper host. Is there a way to command line in Windows?

SkyShadex, 2022-11-26:
[found my answer](https://subscription.packtpub.com/book/hardware-and-creative/9781783284979/1/ch01lvl1sec14/running+slic3r+from+the+command+line+(become+an+expert)).