supermerill/SuperSlicer #3459: Please add a relaease page here or install section

masterleo, 2022-11-28:
on the readme there is no release for mac or linux, it is a bit strange. 
I would like a release sheme like all softwares as there  seems to be many users of this. 
A lot of voron owners are like "please install super slicer"..
as this becomes more mainstream a lot of people will be discouraged because of the "compile it yourself method" 
or did I miss something ? 
Thanks a lot to all developers of this software as this is a brilliant software a lot people like. <3

neophyl, 2022-11-28:
There are prebuilt versions on many of the releases so no need to compile it yourself.  For example https://github.com/supermerill/SuperSlicer/releases/tag/2.4.58.5  has assets for Mac and linux as well as Windows.

Many of the beta's and early test versions don't have mac/linux ones though so for those you might need to compile.  Although even for those you can check https://github.com/supermerill/SuperSlicer/actions  and see if theres been a build for your particular OS.

masterleo, 2022-12-12:
I thank you a lot for this reply. 
I don't understand why I do not see this section here : 
[image]
![Uploading image.png…](

Is the releases folder automaticaly placed on the right now ? I did not see it :(

closed 2022-12-12 by masterleo