supermerill/SuperSlicer #3460: App crashes when started

big-fudge, 2022-11-28:
**Bug description**
Late 2.4 (2.4.58.5) Crashes immediately after start
**Expected behavior**
App start without crashing

!!>> Project File <<!!
-

**To Reproduce**
Start the app

**Screenshots**
-
**Desktop (please complete the following information):**
MacOS 12.6.1 on 2018 MBP 15", 2,2 GHz 6-Core Intel Core i7

baotista, 2022-12-10:
Same here on 12.5.1 2015 MBP 15", 2,2 GHz 4-Core Intel Core i7

galluslongus, 2022-12-10:
Same here on 2.4.58.5
MBP 15" 2018 2,6 GHz 6-Core i7 32 GB
macOS Monterey 12.6.1

Jojo90000, 2022-12-19:
Same on my side, unfortunately... :'(
That, plus the fact I didn't find a good profile for my new FLSun V400... I had to go back (temporary, I hope !!!) on Cura ! Pouah ! :x

Tested on MacBook Air, core i5 (on Catalina & BigSure).
MacBook Pro, core i5 (Catalina)
And Mac Mini, core... well i5 too ! :D (Catalina)

I had to re-install the 2.3.57 release, which works like a charm !

m600x, 2022-12-23:
Fix from https://github.com/supermerill/SuperSlicer/issues/3298#issuecomment-1272537564 work: `brew install zstd`