supermerill/SuperSlicer #3463: [REQUEST] Extruder-specific maximum volumetric speed

Merlin246, 2022-11-28:
Hello,

Would like to request a change to an already-existing feature: "Maximum Volumetric Speed".

I would like to request that either in all cases or at a minimum when the printer config has _extruders_count_ >1, the maximum volumetric speed feature is set in the extruder-specific settings page.

I have a Toolchanger with V6's and volcanos, these have very different maximum flow rates and I would like to be able to optimize print times by making use of that.

Hope this isn't a huge problem to implement :)

dantetango, 2022-11-28:
> Hello,
>
> Would like to request a change to an already-existing feature: "Maximum Volumetric Speed".
>
> I would like to request that either in all cases or at a minimum when the printer config has [...]

I think this is a good suggestion, and I offer a temporary workaround I use- Create a filament profile for each EXTRUDER, and set the max volumetric to that extruder. Just copy your filament profile, and save it with the different max flow.

Merlin246, 2022-11-28:
@dantetango that's a clever workaround.