supermerill/SuperSlicer #3464: Complete individual objects, but if higher than the gantry, split into max gantry height segments

Lecso11, 2022-11-29:
**Is your feature request related to a problem? Please describe.**
I would like to use the _complete individual object_ option a lot more, but right now it is only good for smaller objects than the gantry height.

**Describe the solution you'd like**
I would like to see the slicer to slice it that way, that the printer completes the _object 1_ to the point where it fit below the gantry, and then jumps to _object 2_, completes until fits under the gantry, jumps back to _object 1_ and repeats this until the print completes. Less travel, less stringing, less print time.

Lecso11, 2022-11-29:
Just found #3445. That is pretty much what I need.

closed 2024-02-12 by Lecso11

supermerill, 2024-02-12:
yes, also moving it to the printer / extruder tab.