supermerill/SuperSlicer #3476: When Support is enabled the number of total layers of the object is augmented

Fyr42, 2022-12-03:
### What happened?

On many objets, if I enable the build of support, the numbers of layers is augmented. If the layer is 0.2 on support zone it's ~0.1. The support is build at 0.2. When scrolling the build you can see 2 layers of the object for one layer of the supprt.  

Without support : 160 layers
[image]

Snug Style : 210 layers
[image]

Grid style : 244 layers
[image]

### Project file & How to reproduce

[Cableholder_Ender-3_S1.3mf.zip](https://github.com/supermerill/SuperSlicer/files/10145874/Cableholder_Ender-3_S1.3mf.zip)

Enable or disable Support in the option

### Version

2.4.58.5

### Operating system

macos 13.0.1 (22A400)

### Printer model

Ender 3 S1 Pro (used the non Pro profile)

neophyl, 2022-12-03:
The layer height of the support is completely independent of the object. It always has been. It does not effect the object itself, just the support, but it does show in the preview layers slider.  It’s the same in Prusa Slicer too.  This is normal intended function and not a bug.

Fyr42, 2022-12-03:
So if it's independent why enabling Support changes the number and thickness of the layers of the object  ?

neophyl, 2022-12-03:
It doesn’t. Have you switched the preview of the slice to layer height and checked ?  Just because the layer count changes that’s just the extra ones from the supports. 
Please post your zipped up project file (file>save project as ) to save the 3mf and then zip it up and attach.  It’s the only way to debug things.

Fyr42, 2022-12-04:
project zip already uploaded : [Cableholder_Ender-3_S1.3mf.zip](https://github.com/supermerill/SuperSlicer/files/10145874/Cableholder_Ender-3_S1.3mf.zip)

I took 2 screenshots in layer view , height view, with the color scale near the object for better evaluation of the tint: 
- without support : 0.2 mm
- with support style grid :  0.194 mm (-3%) "0.2 mm" value is no more in the scale. Total size 31.80 (+0.2 mm)
- with support style snug : 0.19 mm (-5%) 

1 - Without support :  0,20 mm
[image]

2 - With support grid : 0.194 mm
[image]

3 - with support snug : 0.19 mm
[image]

Fyr42, 2022-12-04:
For screenshots upper : assuming that the height scale is not decorative, taking into account the regular space between the colors.

I will take some pictures of objects showing the banding where the support is present. It'd not that uniform like in preview.

Fyr42, 2022-12-04:
If you use the slider where there is support, the object had more layers per mm.

Like between 19-20 or 30-31 : normal
But between 0-9 and 20-28 : more layers per mm.

On the no support slider : no variation of ticks per mm.

And the banding is visible.

neophyl, 2022-12-04:
I've now sliced your model and there doesn't appear to be any layer height changes in the model no matter if support is enabled or not.  The colour is uniform throughout and matches the key (although the key values are altered when there is support naturally).  The only difference is the thick bridges as you have nozzle diameter set for that so normal.

Furthermore looking at the actual gcode shows that all the object layer Z heights are on multiples of 0.2.
Travel and wipe moves do adjust the Z height during their operations but the actual model printing moves are back on 0.2 increments.
There is no issue here.