supermerill/SuperSlicer #3483: PNG File Error on start

Mosher23, 2022-12-06:
### What happened?

SuperSlic3r just crashed during slicing and now gives me an persistent error every time on start.

### Project file & How to reproduce

[image]

### Version

2.5.59

### Operating system

Mac OS 12.6.1

### Printer model

Voron V2.4

Evilmonkeybot, 2023-02-08:
I have the same error. I think it's to do with the Voron profile image of the print bed, and the Mac release. After several Force Quits it will load again.
[image]

Version
2.5.59

Operating System
Mac OS 13.2 on ARM

Printer model
Voron V0.1

allenrowand, 2023-03-08:
Problem persists in version 2.5.59.2. Have attached the png image that I'm using for the bed image.
[image]

comatose-tortoise, 2024-02-17:
Samme error on 2.5.59. I can't change the settings, and there's nothing wrong with the image itself. How can I get around this?

supermerill, 2024-02-20:
seems same issue as #4095
I don't really understand what's wrong.
You say it may be the voron build plate? `bedtexture-SW-250x210.png` ?

supermerill, 2024-02-21:
i've output a debug build [here](https://github.com/supermerill/SuperSlicer/actions/runs/7973682768)
It should output error message in the terminal.
can someone try it and give me the output? (It won't resolve the issue, it's just to try to pinpoint it)