supermerill/SuperSlicer #3485: Ridiculous gap fill lines

thwerks, 2022-12-07:
### Version
2.4.58.5

### Operating system type + version
Debian 11

### Behavior
i dont see any way to adjust the threshold at which an internal perimeter gets treated as a gap fill. In my case, what could be 2 or 3 internal perimeters gets treated as a giant gap fill. This doesn't work at all,  speed/temps have to be dramatically altered to accommodate this giant increase in flow rate which itself causes all kinds of undesired effects. Should't there be a way to set the max width at which a internal perimeter gets treated as a gap fill line? probably better the user has control over this threshold.

I noticed that if i set the "overlapping external perimeter" "also for perimeter" to above %50, it actually gives me the result i want but then has a bug where the sparce infill completely overlaps the perimeter so that's even worse 

[image]

[image]

[image]

[Clamp.txt](https://github.com/supermerill/SuperSlicer/files/10171455/Clamp.txt)

CCS86, 2022-12-10:
Read up on this issue I submitted. We had some good discussion: https://github.com/supermerill/SuperSlicer/issues/2967

The wide gap fills are never 2x or 3x your line width, but they can definitely be 1.9x

I run my "also for all perimeters" at 2%, because it is not flow compensated. Basically, whatever value you put there, is an allowable amount of over-extrusion. With parts like yours (and a lot of mine) this can be long distances of over-extrusion and that is terrible for print quality.

Merill was able to add another feature that helps with this in the meantime:
[image]

By using this, those giant width gap fills are slowed way down to keep flow rate in check, and generally print okay for me. You'll notice I keep it around 85%, to set flow rate below a normal perimeter and give it more time to improve bonding.

thwerks, 2022-12-28:
I really appreciate this reply and this does partially mitigate some concerns, but mainly i really want to be able to control the behavior overall. I want to be able to decide the threshold at which gapfill supersedes internal perimeters.  Large gap fill lines are not as strong as smaller lines are are more likely to leak under pressure (such is my concern).

So i want to express my concern and necessity here for this feature, hopefully someone will understand. It does seem like this was at least partially implemented with the "overlapping external perimeter" and "also for perimeter" but its just doesnt work properly, seems like it just needs a bug fix.