supermerill/SuperSlicer #3498: Overhang Perimeters

ckiefer, 2022-12-11:
### What happened?

Superb work in this repo! Thanks to everybody for this great software!

I am pretty sure this is not a bug of the slicer but a problem of the user (me!) not knowing what to do with it ...

**What is the problem?**
I am trying to print this model: https://www.thingiverse.com/thing:3617784 
No matter what I try I always get overhang perimeters in mid-air:
[image]

Is there anything I can do about it or is this a problem of the model? I am not a fan of supports as the overhangs also appear further up in the model:
[image]

### Project file & How to reproduce

[Death_Planter.zip](https://github.com/supermerill/SuperSlicer/files/10202598/Death_Planter.zip)

### Version

Version 2.5.59 (Arachne nightly build)

### Operating system

windows 10

### Printer model

Anycube MZ 2.0

neophyl, 2022-12-11:
The issue is with the model design.  The first image you posted has a downward sloping bottom surface, as such the outer perimeter for it is printed in  midair.  Thats exactly why supports are needed.  And its going to need a lot of them.  The thingiverse entry even says supports are needed.  Read the comments on the thing and you will see people having trouble even with supports.  Its just not well designed for fdm printing.

abrasius, 2023-06-16:
Got this interesting slicing result with runaway external perimeter:
[image]

..And got it to behave by adding curve smoothing of 0.1mm:
[image]