supermerill/SuperSlicer #3504: Bed Texture blacked out

JustinQuintero, 2022-12-14:
### Version
_2.4.58.5_

### Operating system type + version
_Windows 10 (build 22623), 64-bit edition_

### Behavior
* _When a custom bed texture is loaded in "Printer Settings --> General --> Bed Shape" the texture is blacked out in the 3D view_
* _Steps needed to reproduce the problem: Load a custom texture_
* _May be it is an AMD related graphics issue? as I share the same settings with an old Windows 10 Laptop I have which has an NVIDIA GPU and the texture works on that device. My system is running an AMD 6700XT GPU and AMD 5900X CPU._

[image]

neophyl, 2022-12-15:
Please see https://github.com/prusa3d/PrusaSlicer/issues/8417

margt1974, 2022-12-21:
I have found it - that the slicer has the problems with .PNG file. If you convert the file in .SVG - problem solved.
Once - you must load this new in the settings.
The .PNG files are in SuperSlicer files: resources - profiles - Voron (example)

JustinQuintero, 2023-01-17:
Found that if you resize the .png to either 512x512 or 1024x1024 it displays correctly