supermerill/SuperSlicer #3512: File name formatting

carlbeech, 2022-12-18:
**Is your feature request related to a problem? Please describe.**
Not related to a problem

**Describe the solution you'd like**
(Apologies if I've missed where this is - I've started to use Superslicer only recently)
Would it be possible in preferences (Files section) to put in an output file name format 

**Describe how it would work**
Simple keyword substitution (I believe its already taking place, but can't find it in the GUI/ not sure where to look in INI files) [base name][print time][FilamentUsed_G] etc....   main things I'm usually interested in are time taken, and filament amount in grams.

**Describe alternatives you've considered**
I just have to keep updating the file name with the amounts I see from the screen.

Many thanks

Carl.

carlbeech, 2022-12-18:
Sorry - located ....

Its under print settings - however, I'm not sure for the reasoning - I would have thought that you'd want to set once (i.e. why I was looking in preferences?) for all situations - whereas under 'print settings' it would appear that you can have / have to update for each profile...

Guess I've not fully got to grips with some of the basics... :-)

Now, just trying to work out the correct keywords to locate filament used in grams...

carlbeech, 2022-12-18:
Ah - spotted 'used_filament' - however how to convert to grams?

An actual suggestion would be a variable constructor...   basically a pulldown that drops a relevant variable in the cursor point... perhaps with variables having some form of basic description to make things easier?

WickedInvi, 2022-12-19:
Each field contains the variable name that needs to be used when you hover over it. Perhaps a right-click option to "add to output file name" can be added.

carlbeech, 2022-12-19:
Hi 
Thanks - yep - thats where I found the used_filament tag.. :-)

Interesting - I would have thought filament used in grams would be something that's fairly basic...  I've always used the same filament supplied (ESun) - and all the spools are the same...   when you weigh an empty spool and take this away from the total for spool and filament, then you know how many grams are left...  so if you're doing a bigger print - you don't have to guess if you've got enough filament left on the spool...

I've taken the concept a little further at home - I designed a dessicant holder, which fits in a box which holds a spool to keep it dry, and it has a hole for filament to come out - and kept reasonably airtight by flowing through sponge...  the thing is, I weigh it so that empty spool + dessicant + other bits of the box = 1Kg...  so when I weigh the whole thing (not having to take out of the airtight box), I know how much filament is on the spool...

Hope that makes sense...   

At the end of the day, 'Sliced info' has Used Filament (g) right next to Used Filament (m)... so I wouldn't have thought it that hard to have as a variable for file names....

neophyl, 2022-12-19:
the weight is one of the basic ones, has been in there forever too.  its total_weight.  although in normal use the issue is its to many decimal places.  You can fix that though by using "...{int(total_weight)}g..."
For more decimals use "...{int((total_weight*10.0))/(10.0)}g..." or replace the "10.0" with any order of magnitude you like.

In the end for mine I decided that I just wanted to the value as a rough guide to make sure I had enough filament left so I now just use _{int(total_weight)+1}g which always puts it up above the actual value and nicely rounded off.

carlbeech, 2022-12-19:
Can't believe I missed that! - many thanks for pointing it out :-)    - very useful with the extra decimals function - I'll be implementing that :-D

carlbeech, 2022-12-19:
Much appreciated to all :-)