supermerill/SuperSlicer #3517: Application fails to launch

MoldyHacker, 2022-12-20:
### What happened?

I have been using v2.4.58.5.
I used the program a week ago, and it worked just fine. Recently the program has been failing to launch. When I launch it, the program hangs and is unresponsive. I downloaded a fresh copy which didn't fix the problem. I tried an older version, but that didn't fix the problem either. I believe this may have something to do with a windows update, but I can't put my finger on what.

Edition	Windows 10 Pro
Version	22H2
Installed on	‎5/‎17/‎2022
OS build	19045.2364
Experience	Windows Feature Experience Pack 120.2212.4190.0

### Project file & How to reproduce

Failure to launch

### Version

2.4.58.5

### Operating system

Windows 10

### Printer model

prusa i3 MK2 MMU

Allidoisowen, 2022-12-22:
I just had a similar issue.
After testing I found it was a corrupt print setting preset.

Magoo0876, 2022-12-23:
Hi, I also can not use SS anymore. After the update to the latest version, there is a problem with : 
Library not loaded: /usr/local/opt/zstd/lib/libzstd.1.dylib

Process:               SuperSlicer [8408]
Path:                  /Applications/SuperSlicer.app/Contents/MacOS/SuperSlicer
Identifier:            com.merill.slic3r/
Version:               SuperSlicer SuperSlicer_2.4.58.5_2022-09-23 (SuperSlicer_2.4.58.5_2022-09-23)
Code Type:             X86-64 (Native)
Parent Process:        ??? [1]
Responsible:           SuperSlicer [8408]
User ID:               501

Date/Time:             2022-12-23 10:52:25.105 +0100
OS Version:            macOS 11.6.2 (20G314)
Report Version:        12
Bridge OS Version:     6.1 (19P647)
Anonymous UUID:        7FAA0A14-A2D9-27F8-25BE-85B977437627

Time Awake Since Boot: 98000 seconds

System Integrity Protection: enabled

Crashed Thread:        0

Exception Type:        EXC_CRASH (SIGABRT)
Exception Codes:       0x0000000000000000, 0x0000000000000000
Exception Note:        EXC_CORPSE_NOTIFY

Termination Reason:    DYLD, [0x1] Library missing

Application Specific Information:
dyld: launch, loading dependent libraries

Dyld Error Message:
  dyld: Using shared cache: C6E05320-A8C1-395E-9193-122E5CB33FD0
Library not loaded: /usr/local/opt/zstd/lib/libzstd.1.dylib
  Referenced from: /Applications/SuperSlicer.app/Contents/MacOS/SuperSlicer
  Reason: image not found

Binary Images:
       0x105b6f000 -        0x107c42fff +com.merill.slic3r/ (SuperSlicer SuperSlicer_2.4.58.5_2022-09-23 - SuperSlicer_2.4.58.5_2022-09-23) <EA3F7209-F390-3268-9C90-D8081DCEDC30> /Applications/SuperSlicer.app/Contents/MacOS/SuperSlicer

MoldyHacker, 2022-12-26:
I found that running as administrator on windows fixes the not responding problem.