supermerill/SuperSlicer #3521: Include option to change where SS looks for appdata.

Allidoisowen, 2022-12-22:
**Is your feature request related to a problem? Please describe.**
I noticed pull request Some [Windows/RDP related issues fixes #3443] and reminded me of an issue.
I run an RDP session for slicing as my workstations are pi's used for RDP back to my old xeon with a k2200. I can slice while anywhere on anything.
Sometimes on my main rig while modeling, I want to slice and print on my prototype printer next to me. I use the local SS install.
If I made a change to anything in my main environment, those changes are not carried over to my personal rig. I rarely remember to copy configs before at least one print.

And please merge the RDP fix if it works, thanks!

**Describe the solution you'd like**
I would like to be able to host the config files in a central location. 

**Describe how it would work**
The common files stored in the appdata\roaming\superslicer would be hosted on a mapped network drive/drive share/or any other non-local location.
Default run behavior would not change. 
However users would have the ability to point the application to an alternate location for the .ini files.
I would image this setting could be stored in the current location in a small file that would call the external files.

**Describe alternatives you've considered**
I've thought about a type of symbolic link or file that would point toward a common store of files.
I figured it would be slow to load and cumbersome to maintain should any new files be added in the future.

**Additional context**
That's all I have to say.

Also, if the experience over RDP or similar be improved it would help greatly in standing up multiple low cost solutions for workstations while maintaining a decent slicing environment. 

Thanks for all your hard work!!!

ssorgatem, 2022-12-22:
Can't you just make a symbolic link and redirect the appdata\roaming\superslicer to a network drive?
It wouldn't be any slower than without the symbolic link.

If you want to avoid symbolic links for some reason, another option would be to launch SuperSlicer but setting the %APPDATA% variable to point to the network drive. That way SuperSlicer will store and look for its configuration there.
On Linux it'd be the XDG_CONFIG_HOME variable.

Allidoisowen, 2022-12-22:
> If you want to avoid symbolic links for some reason, another option would be to launch SuperSlicer but setting the %APPDATA% variable to point to the network drive.

That's a good idea. I didn't know that was available. Thank you for the tip!

julianjc84, 2022-12-22:
If you make a 'configuration' folder next to the .exe the app will  use that over appdata location. (this should be default behaviors)
You can even run multiple instances/versions of superslicer (when you duplicate the app like this - great for different printer) with each having its own 'configuration'.
It will even run from a network drive /googledrive/onedrive on multiple machines. You only have to worry about making changes at same time on multiple machines.

https://github.com/supermerill/SuperSlicer/discussions/1772#discussioncomment-1577050
[image]