supermerill/SuperSlicer #3522: The tab "Single extruder MM setup" does not show up anymore, neither on release nor on nightly build

kursatu, 2022-12-23:
**Bug description**
This is from Prusa slicer.
[image]
The options for enabling it, is checked.
[image]

**Expected behavior**

**Desktop (please complete the following information):**
 - OS: Windows
 - Version: LAte 2.4 and the build I've made from the sources.
 - Printer model

**Additional context**
Add any other context about the problem here.

spikeygg, 2023-01-10:
BeOddTodd just showed me that you can de-select and re-select the "Single Extruder Multi Material" option in General to make the "Single extruder MM setup" option appear in version 2.4.58.5. Apparently, that value doesn't get read properly on application launch but you can work around it by de-select/re-selecting the option.

LAP87, 2023-03-25:
Seems to be fixed now

spikeygg, 2023-03-25:
> Seems to be fixed now

Nope, it is not fixed, yet.

Here's a animated GIF of 2.5.59.2 (current release) with the issue and how to "work around" it:
[image]