supermerill/SuperSlicer #3524: MacOS library missing

Biclops, 2022-12-24:
### What happened?

Open SuperSlicer_2.4.58.5_macos_220924.dmg on a Intel based Mac.

You'll see the following:

`Process:               SuperSlicer [23497]
Path:                  /Applications/SuperSlicer.app/Contents/MacOS/SuperSlicer
Identifier:            com.merill.slic3r/
Version:               SuperSlicer SuperSlicer_2.4.58.5_2022-09-23 (SuperSlicer_2.4.58.5_2022-09-23)
Code Type:             X86-64 (Native)
Parent Process:        ??? [1]
Responsible:           SuperSlicer [23497]
User ID:               501

Date/Time:             2022-12-24 13:41:22.464 -0800
OS Version:            Mac OS X 10.15.7 (19H2026)
Report Version:        12
Anonymous UUID:        FB12F534-BF6E-00F2-683F-14B75D68CBDB

Time Awake Since Boot: 480000 seconds

System Integrity Protection: enabled

Crashed Thread:        0

Exception Type:        EXC_CRASH (SIGABRT)
Exception Codes:       0x0000000000000000, 0x0000000000000000
Exception Note:        EXC_CORPSE_NOTIFY

Termination Reason:    DYLD, [0x1] Library missing

Application Specific Information:
dyld: launch, loading dependent libraries

Dyld Error Message:
  Library not loaded: /usr/local/opt/zstd/lib/libzstd.1.dylib
  Referenced from: /Applications/SuperSlicer.app/Contents/MacOS/SuperSlicer
  Reason: image not found

Binary Images:
       0x10a070000 -        0x10c143fc7 +com.merill.slic3r/ (SuperSlicer SuperSlicer_2.4.58.5_2022-09-23 - SuperSlicer_2.4.58.5_2022-09-23) <EA3F7209-F390-3268-9C90-D8081DCEDC30> /Applications/SuperSlicer.app/Contents/MacOS/SuperSlicer
       0x11d0eb000 -        0x11d17d267  dyld (750.7) <9F48F7F8-94D0-3793-99B7-DDEF657EF956> /usr/lib/dyld
    0x7fff32407000 -     0x7fff32887ff9  com.apple.CoreFoundation (6.9 - 1678.104) <CA9600EB-C7BE-38C4-968D-4BF90B1C9550> /System/Library/Frameworks/CoreFoundation.framework/Versions/A/CoreFoundation
    0x7fff35225000 -     0x7fff352c9ff3  com.apple.framework.IOKit (2.0.2 - 1726.148.1) <0A9D244C-AECC-3D29-86DE-4F3B04F8DD25> /System/Library/Frameworks/IOKit.framework/Versions/A/IOKit
    0x7fff3cde7000 -     0x7fff3cdf6ff7  com.apple.opengl (17.10.22 - 17.10.22) <BCC1E523-1843-3FF8-8F7D-29DA67756434> /System/Library/Frameworks/OpenGL.framework/Versions/A/OpenGL
    0x7fff68513000 -     0x7fff68514fff  libSystem.B.dylib (1281.100.1) <5A20AE6C-BF4D-3689-B7D2-19F0721F6375> /usr/lib/libSystem.B.dylib
    0x7fff68882000 -     0x7fff6888efff  libbz2.1.0.dylib (44) <14CC4988-B6D4-3879-AFC2-9A0DDC6388DE> /usr/lib/libbz2.1.0.dylib
    0x7fff6888f000 -     0x7fff688e1fff  libc++.1.dylib (902.1) <59A8239F-C28A-3B59-B8FA-11340DC85EDC> /usr/lib/libc++.1.dylib
    0x7fff69385000 -     0x7fff69475fff  libiconv.2.dylib (59) <18311A67-E4EF-3CC7-95B3-C0EDEE3A282F> /usr/lib/libiconv.2.dylib
    0x7fff696e9000 -     0x7fff69701ff3  liblzma.5.dylib (16) <C131EF18-2CDD-3271-8A30-A8760D4FE166> /usr/lib/liblzma.5.dylib
    0x7fff6aadf000 -     0x7fff6aaf1ff3  libz.1.dylib (76.0.1) <E3495313-4CFF-34C4-AD9F-5DE24F66E15D> /usr/lib/libz.1.dylib

Model: Macmini6,2, BootROM 429.0.0.0.0, 4 processors, Quad-Core Intel Core i7, 2.3 GHz, 16 GB, SMC 2.8f1
Graphics: kHW_IntelHD4000Item, Intel HD Graphics 4000, spdisplays_builtin
Memory Module: BANK 0/DIMM0, 8 GB, DDR3, 1600 MHz, 0x0D9B, -
Memory Module: BANK 1/DIMM0, 8 GB, DDR3, 1600 MHz, 0x0D9B, -
AirPort: spairport_wireless_card_type_airport_extreme (0x14E4, 0x10E), Broadcom BCM43xx 1.0 (7.21.190.33 AirPortDriverBrcm4360-1601.1)
Bluetooth: Version 7.0.6f8, 3 services, 27 devices, 1 incoming serial ports
Network Service: Wi-Fi, AirPort, en1
Network Service: iPhone 2, Ethernet, en5
Network Service: iPhone 3, Ethernet, en6
Serial ATA Device: APPLE SSD SM256E, 251 GB
USB Device: USB 3.0 Bus
USB Device: USB3.0 Hub
USB Device: USB3.1 Hub
USB Device: USB3.1 Hub
USB Device: USB2.0 Hub
USB Device: USB2.1 Hub
USB Device: USB2.1 Hub
USB Device: iPhone
USB Device: USB Receiver
USB Device: Mass Storage Device
USB Device: USB 2.0 Bus
USB Device: Hub
USB Device: Hub
USB Device: IR Receiver
USB Device: BRCM20702 Hub
USB Device: Bluetooth USB Host Controller
USB Device: USB 2.0 Hub
USB Device: Composite Device
USB Device: USB 2.0 Bus
USB Device: Hub
Thunderbolt Bus: Mac mini, Apple Inc., 23.4
`

### Project file & How to reproduce

N/A

### Version

2.4.58.5

### Operating system

macOS Catalina 10.15.7 (19H2026)

### Printer model

Ender 3

Seepirat, 2022-12-25:
I see the same issue with macOS Monterey Version 12.2.1.

cihadaydemir, 2022-12-25:
Same issue on Ventura 13.0.1

Stout001, 2022-12-25:
Same issue on Mojave 10.14.6 and no workaround.

Seepirat, 2022-12-26:
I installed the zstd lib via Homebrew on my iMac and now super slicer is running.
You could install the zstd lib with following command in the terminal:
brew install zstd

Stout001, 2022-12-26:
> brew install zstd

Thanks,
However it does not seem to work on Mojave. Complaining about too old OS. I am not prepared to upgrade from Mojave as my dual monitors stop working in later MacOS

nanoplane, 2022-12-31:
So.... the build that is being used to create this version builds in an environment that has zstdlib installed via home-brew and for some reason, the build doesn't include the library in the distribution.  

If this code is built in an environment where zstdblibj has NOT been installed... everything works fine... (I do it with my fork all the time and it never needs zstdlib to build and run on the mac.

Henk72, 2023-01-02:
> > brew install zstd
> 
> Thanks, However it does not seem to work on Mojave. Complaining about too old OS. I am not prepared to upgrade from Mojave as my dual monitors stop working in later MacOS

For met it worked on Mojave with the brew install zest

RobinWts, 2023-01-05:
Same here (MacOS 12.6.2)
`Termination Reason:    Namespace DYLD, Code 1 Library missing
Library not loaded: '/usr/local/opt/zstd/lib/libzstd.1.dylib'
Referenced from: '/Applications/SuperSlicer.app/Contents/MacOS/SuperSlicer'
Reason: tried: '/usr/local/opt/zstd/lib/libzstd.1.dylib' (no such file), '' (no such file), '' (no such file)
(terminated at launch; ignore backtrace)
`

jimwhitelaw, 2023-01-17:
Confirming another: 

```
Process:               SuperSlicer [16784]
Path:                  /Applications/SuperSlicer.app/Contents/MacOS/SuperSlicer
Identifier:            com.merill.slic3r/
Version:               SuperSlicer SuperSlicer_2.4.58.5_2022-09-23 (SuperSlicer_2.4.58.5_2022-09-23)
Code Type:             X86-64 (Native)
Parent Process:        launchd [1]
User ID:               501

Date/Time:             2023-01-17 10:35:28.0136 -0700
OS Version:            macOS 12.6.2 (21G320)
Report Version:        12
Anonymous UUID:        1D05FE3F-4795-E817-E0FF-91FCFDA93368

Time Awake Since Boot: 920000 seconds

System Integrity Protection: enabled

Crashed Thread:        0

Exception Type:        EXC_CRASH (SIGABRT)
Exception Codes:       0x0000000000000000, 0x0000000000000000
Exception Note:        EXC_CORPSE_NOTIFY

Termination Reason:    Namespace DYLD, Code 1 Library missing
Library not loaded: '/usr/local/opt/zstd/lib/libzstd.1.dylib'
Referenced from: '/Applications/SuperSlicer.app/Contents/MacOS/SuperSlicer'
Reason: tried: '/usr/local/opt/zstd/lib/libzstd.1.dylib' (no such file), '' (no such file), '' (no such file)
(terminated at launch; ignore backtrace)

```

jimwhitelaw, 2023-01-17:
Further, I have confirmed that the issue is not present in 2.3.57.11. It started with the first 2.4 release.