supermerill/SuperSlicer #3527: All Filaments missing

JanB97, 2022-12-25:
**Bug description**
I tried to add a custom variable (printer_name) to my printer and add it to my file name.
After slicing I got a access violation error of some sort and needed to restart superslicer.
Now all my filaments are gone, it only shows "default" as filament.
My print profiles are still there.
In appdata I can see my filaments ini files and I can import them back, but now after every restart of the software they are gone again.
[image]

Here is a crash report I found.
[crash_log.txt](https://github.com/supermerill/SuperSlicer/files/10300759/crash_log.txt)

**Expected behavior**
Filaments in appdata folder should load.

**To Reproduce**
Steps to reproduce the behavior:
Na, maybe add the variable to the printer?

**Desktop (please complete the following information):**
 - OS: Windows 10 22h2
 - Version: 2.4.58.5
 - Printer model: Voron Trident 300 - Klipper

JanB97, 2022-12-25:
Fixed it myself. 
It had some kind of problem with the printer.ini.
Copied the content to a new file, deleted the old one and renamed. 
Now it is working again. Weird.

closed 2022-12-25 by JanB97

JanB97, 2022-12-26:
For anyone finding this, I am using a Voron and have the default Voron2 0.4 profile.
If I change any of the notes of the printer (even on a copied version of it) it breaks.

inline6power, 2023-05-20:
> Fixed it myself. It had some kind of problem with the printer.ini. Copied the content to a new file, deleted the old one and renamed. Now it is working again. Weird.

you are a life saver! thanks a ton. i had this same thing happen and i had a bunch of filaments saved with different pa values. thanks a bunch

JanB97, 2023-05-24:
> you are a life saver! thanks a ton. i had this same thing happen and i had a bunch of filaments saved with different pa values. thanks a bunch

Your welcome, that's why I left it here.
I hate when I find people with the same problems and they just write "fixed it" without giving the solution:)

I since found out what the actual "problem" is and it's more the intended behavior.
For the Voron profile at least, all generic filaments and therefore all copies have a dependency that searches the printer description for "E3D V6" or something similar, I don't actually know the exact term anymore.
If you remove this from the printer settings, all filaments won't meet that dependency anymore and will therefore not be shown anymore.
I think there should be a options to show all existing filaments no matter the dependency, in cases where you screw up, but anyways...