supermerill/SuperSlicer #3532: Error slicing, temp file locked?

Etienneg93, 2022-12-28:
### Version
_Version of SuperSlicer used goes here (help->about)_
2.5.59.0
### Operating system type + version
_What OS are you using, and state any version #s_
Windows 10 Home
Version 22H2

### Behavior
* _Describe the problem_
Can't slice any file, always get the same error 

Failed to rename the output G-code file from C:\Users\rozzo\AppData\Local\Temp\.16524.gcode.tmp.postprocess to C:\Users\rozzo\AppData\Local\Temp\.16524.gcode.tmp
Is C:\Users\rozzo\AppData\Local\Temp\.16524.gcode.tmp.postprocess locked? 

* _Steps needed to reproduce the problem_
Slice any file
* _Expected Results_
Get a fully sliced gcode file
* _Actual Results_
Error: Failed to rename the output G-code file from C:\Users\rozzo\AppData\Local\Temp\.16524.gcode.tmp.postprocess to C:\Users\rozzo\AppData\Local\Temp\.16524.gcode.tmp
Is C:\Users\rozzo\AppData\Local\Temp\.16524.gcode.tmp.postprocess locked? (gcp)permission denied
[image]
  * _Screenshots if useful_

lrockz, 2023-01-19:
Hello I am experiencing the same issue however only with one file. The file I am trying to slice/print is from an project of my own from Fusions360 that I export as mesh and then attempt to print. 
After slicing the STLs I get prompt with the same error. Depending on the profile/print settings used I also get prompt with another error, the "Access Violation" error. 

The files are mines and should be exported correctly, I have printed those STLs in the past from previous versions of the same project without issues, so I don't know what is happening. 

If required I can share some logs, let me/us know if this is easy to fix.

Etienneg93, 2023-01-19:
I believe this may be related to windows updates enabling antivirus?
Real-time protection seems to have put my entire APPDATA folder into read-only mode

lrockz, 2023-01-20:
My Appdata folder also seems to be on read-only mode but I doubt this has to do with anything since I'm able to slice other items fine. I did attempt to disable real-time protection and run SS as Admin to see if it would help but no luck there.

I will attach my crash log from this morning
[crash_log.txt](https://github.com/supermerill/SuperSlicer/files/10463986/crash_log.txt)

lrockz, 2023-01-22:
So I've attempted to create a new Fusion360 project and redo the print. Slice the new STL, and get the same error... Tried slicing the same files with PrusaSlicer and it works fine. Really don't know what is causing this