supermerill/SuperSlicer #3535: Bridge infill comes in at unwanted places, with no option to disable it.

Hamrbooo, 2022-12-28:
**Is your feature request related to a problem? Please describe.**
When slicing models with overhangs, SuperSlicer (version 2.4.58.5) creates this thing called 'Bridge infill' in the sliced model. The problem is that even though it's called infill, it mostly generates on the **outside** of the mesh. It doesn't stop there, because it can also generate inside other perimeters, making the extruder try to extrude filament in a place where it already extruded filament. This can be seen when slicing a benchy (which I got from here https://www.thingiverse.com/thing:763622):
[image]

A Benchy has some inlayed text on the bottom, which shouldn't be filled with filament. In the screenshot above you can see SuperSlicer filled it with Bridge infill (and no, supports are turned off). You can also see the Bridge infill overlapping with the Internal and External perimeters.

To give an example of what I would expect, here is what Cura did:
[image]

You can see Cura didn't fill up those letters.

This issue can also be seen on vertical holes:
[image]
Here SuperSlicer puts all layers nicely around the hole except for the top where it puts a Bridge infill line inside the hole.

Since SuperSlicer is based on PrusaSlicer, I checked how PrusaSlicer handles this. PrusaSlicer 2.4 did the exact same thing with Bridge infill as SuperSlicer 2.4 did. So the same issue exists there. I also checked PrusaSlicer 2.5, but to my surprise it gets even worse:
[image]
It still fills up the text (which is wrong) but it also overlaps with almost the whole bottom of the benchy!

I searched the internet and tried finding a way to turn this bridge infill off, but there is no option to do that. Even setting bridge flow to 0% doesn't work, because 2% is the lowest.

**Describe the solution you'd like**
Ideally, I would like to see he bridge infill getting generated **inside** the mesh and not overlapping with other perimeters. I see why the bridge infill gets generated; bridging is a good way to be able to make horizontal overhangs and the bridge infill does a good job finding the shortest bridging length. It just has to be generated a few layers higher so it ends up inside the model.
I would also accept the option to just turn the bridge infill off. Although this is less ideal since you don't get the power of bridging on overhangs, at least your model will print in the intended shape.

**Describe how it would work**
For the ideal solution, the slicer would take the thickness of the bridge infill into account and makes it so that it doesn't get generated lower than it should. (I found out that the bridge infill has the same diameter as the nozzle, and doesn't have anything to do with the set layer height.) It also shouldn't overlap with other perimeters like it does in PrusaSlicer 2.5.
The other option, just being able to disable it, could be implemented as a checkbox in the settings. Maybe it could also be implemented in the gcode tab:
[image]
Maybe it's possible to have checkboxes in this list, so the user is able to extract a certain feature type from the gcode with one click. Or maybe something like right mouse click -> exclude from gcode would work.

**Describe alternatives you've considered**
The alternative is going back to Cura, but I really don't wanna do that since I like SuperSlicer's UI a lot more. For now I slice my models with SuperSlicer first, and when Bridge infill shows up, then I'll slice it with Cura.

**Additional context**
Sometimes, even Overhang perimeters can get generated lower than they should:
[image]
This isn't that much of a problem, but since it's somewhat similar to what bridge infill can do, I figured I'll just mention it.

lucascarvalho, 2023-03-10:
I have the same issue while printing these https://github.com/VoronDesign/VoronUsers/blob/master/printer_mods/richardjm/snap-latch-2020/STL/NewLatch-3.0mm.stl
[image]

Hamrbooo, 2023-05-22:
> Selecting the settings "Simulate Prusa No Thick Bridge" fixed the issue for me.

This seems to solve the issue for me as well.

closed 2023-05-22 by Hamrbooo

roe3dd, 2024-08-23:
> Selecting the settings "Simulate Prusa No Thick Bridge" fixed the issue for me.

Where do I find this setting? Please thx!