supermerill/SuperSlicer #3540: Flat surface is not aligned (?) to bed

ABLomas, 2022-12-29:
### What happened?

Attached is +/- simple stl file to demonstrate issue.
On one (shorter) side it aligns ("F" -> select face) OK to bed.
On wider side it does not (?):
[image]
when printed - it becomes like created from two parts.
Part made from one solid boss/extrude in solidworks.
Usually flat side is aligned to bed and printed in one go, but this one does not align properly.

### Project file & How to reproduce

[r_tube_rud_mnt.zip](https://github.com/supermerill/SuperSlicer/files/10321960/r_tube_rud_mnt.zip)

### Version

Version: 2.5.59.0

### Operating system

Win10

ABLomas, 2022-12-29:
Also sliced with Simplify3D 4.1.1 - no issues, first layer is in one piece as expected.

ABLomas, 2024-02-18:
Still the same issue exists in "Discrete legend Edition" (and other earlier releases). Also decided to check if printer will do as superslicer shows (maybe it's display issue) - yes, it does:
[image]
Bottom of part was flat with some supports, but flat part was split in two (and later not printed)  like it would be with angled surface.

supermerill, 2024-02-23:
same "bug" in prusaslicer2.5: the rotate-to-plane doesn't rotate your part exactly where it should be.
Seems fixed in 2.7.
Solution 1: rotate it in your modeling software
Solution 2: you can cut the bottom 0.2 mm, so it's flat.