supermerill/SuperSlicer #3546: Thumbnail Color Not Reflecting Filament Color

3dcauldron, 2022-12-31:
### What happened?

When I generate gcode with thumbnails they are always yellow, unless I use the "Color overide" option. 
I have changed the color on the filament tab, and tried several different filaments. 
This seems to be an issue with 2.4.58.5 and 2.5.59
I have tried thumbnails with both octoprint with slicer thumbnail plugin, and fluidd. 

### Project file & How to reproduce

[thumb issue.zip](https://github.com/supermerill/SuperSlicer/files/10327856/thumb.issue.zip)

### Version

2.4.58.5 and 2.5.59

### Operating system

windows 10 pro

### Printer model

ender 3

JanB97, 2023-01-01:
In the default configuration Superslicer uses the Extruder / Toolhead color for thumbnails.
Took me a while to figure out as well.

You need to go to your printer settings -> Extruder (number) -> Preview -> click "reset to filament color" or something like that.

3dcauldron, 2023-01-03:
> 

Yeah I found that. It works after clicking the button. On the general tab under printer settings the tool tip for "color override" states that if not enforce it will be the one defined by filament.  It seems its actually the color defined by the extruder. 
So its either a bug (getting color from wrong place) or the tool tip should be updated.

greenbigfrog, 2023-01-10:
> You need to go to your printer settings -> Extruder (number) -> Preview -> click "reset to filament color" or something like that.

Thanks.

> On the general tab under printer settings the tool tip for "color override" states that if not enforced it will be the one defined by filament. It seems its actually the color defined by the extruder.

Well, it makes sense to have the priority be `extruder > filament > default` in a multi extruder setup I guess.