supermerill/SuperSlicer #3548: Min. Layer Time Speed Reduction Glitch

CCS86, 2023-01-01:
### What happened?

For some reason, when layer time suddenly drops, the first layer isn't speed reduced like it should be:

[image]

You can see here that the first small layer is printed at a significantly higher speed than the identically sized layer above it. This causes a layer time of 5 seconds, when 7 seconds minimum is specified.

### Project file & How to reproduce

[min-layer-time.3mf.txt](https://github.com/supermerill/SuperSlicer/files/10328252/min-layer-time.3mf.txt)

### Version

2.5.59.0

### Operating system

Windows 10 x64

CCS86, 2023-12-24:
@supermerill

supermerill, 2023-12-29:
This is to avoid sudden speed change for just a little layer, when there is a huge one, a little one and then a big one again.
Currently, it can't look at the next one, so i can't be sure.

CCS86, 2023-12-29:
Hmm, maybe a child setting to allow single layer skipping, or an averaging of X number of layers in that calculation. 

I personally don't see why you would use the minimum layer time setting, and then be upset that it slowed down a single layer which fell below the minimum you set.

supermerill, 2023-12-29:
here, you can see the "zebra" in speeds.
https://github.com/supermerill/SuperSlicer/issues/439
I'll add a setting in the future.