supermerill/SuperSlicer #3553: Flow calibration not generating good models

stratiegery01, 2023-01-04:
### Version
_Version of SuperSlicer used goes here (help->about)_
2.4.58.5

### Operating system type + version
_What OS are you using, and state any version #s_
Win64

### Behavior
* _Describe the problem_
* When generating "10% intervals..." of "2% intervals...", the model produced overlaps itself and is not modifiable. 
* _Steps needed to reproduce the problem_
* menu > calibration > filament flow calibration > [selection of intervals]
* _Expected Results_
* the models should be printable 
* _Actual Results_
  * _Screenshots if useful_
  * 
[image]

_Is this a new feature request?_ No
Related guides for writing feature requests: http://meta.stackexchange.com/a/259196 http://nickohrn.com/2013/09/write-great-feature-request-bug-report/

#### Project File (.3MF) where pro
[Flow calibration 10 issue.zip](https://github.com/supermerill/SuperSlicer/files/10341617/Flow.calibration.10.issue.zip)
blem occurs
_*Upload a Project File* (.3MF) `Plater -> Save Project as...`_
_You have to zip it (or rename it to .zip) to be able to drag& drop it here_

awcsorensen, 2023-01-05:
I had this issue when the "preferred orientation" in "printer settings" was set to a non zero value.

stratiegery01, 2023-01-15:
> I had this issue when the "preferred orientation" in "printer settings" was set to a non zero value.

Thanks! that worked,

closed 2023-01-15 by stratiegery01