supermerill/SuperSlicer #3566: What "Locks" Max Print Speed in Print Settings?

inspectionsbybob, 2023-01-09:
### Version
2.4.58.5

### Operating system type + version
_What OS are you using, and state any version #s_

### Behavior
Occasionally the /Print Settings/Max Print speed (Auto Speed Advanced)  
or 
                        /Print Settings/Speed for Print Moves/ Default Speed

Becomes "GREY" or Locked.

As I am constantly "Tweaking" my print speeds this is a pain, and I often have to go back to a different profile that wasn't locked and move all my changes over (again).

So my question is, what is the Hierarchy of the Speed Variables, what locks what?

I want to know what is the "master" that everything else is a percentage of unless a specific speed is given.

Is It Filament Setting/Print Speed Override then Print Settings /Max Print Speed 

The attached 3MF file SHOULD show those as locked, it it doesn't that is what I am fighting, what locks those variables?

_Is this a new feature request?_
Documentation Request
[rotor.zip](https://github.com/supermerill/SuperSlicer/files/10376476/rotor.zip)

#### Project File (.3MF) where problem occurs
_*Upload a Project File* (.3MF) `Plater -> Save Project as...`_
_You have to zip it (or rename it to .zip) to be able to drag& drop it here_

neophyl, 2023-01-09:
Bob I'm not 100% on the following but from a little experimentation I can see the following.....

As soon as you define a Volumetric Speed in Auto Speed then the max print speed is greyed out.  Which does not make sense as you can configure a Max Volumetric under the Filament settings and if you do it does NOT grey out the setting in Print Settings.
Can not figure out the reasoning behind that.

On the other speed for print moves values, well those at least make a bit more sense.  Think of the columns as Primary/secondary and tertiary.  If you have fixed values in the Primary columns then the Default Speed is greyed out.  As none of them are basing their speed on the default value.  
If you set a percentage % based value in them then the default speed is editable as then you have a value that is based on the default speed.  This is mentioned in the tooltip.
The only one of the 'primary' values that doesn't obey this is gapfill.  Probably as that speed is based on the internal Perimeter speed (also mentioned in the tooltip).

The secondary and tertiary columns are based upon their rows primary value so putting a % in those doesn't effect the editable properties of the Default speed box.
There are other logical inconsistencies in the secondary/tertiary availability, like Support Interface is greyed out if supports aren't enabled which is fair enough, but then why isn't Support speed ALSO greyed out if supports are turned off ?  Possibly someone asked for the support speed to be editable even if supports were turned off at some point ?  Who knows.

inspectionsbybob, 2023-01-09:
Thanks...
I think I had a mix of Percentages -AND- absolute values in my secondary column and that locked the Max...

I still think it would be a good idea to put the reason next to anything that is greyed out.  Something like:
[ non-percentage value in speed column locks max speed]

would have taken me forever to figure that one out...

thanks again...

NameOfTheDragon, 2023-01-13:
> The attached 3MF file SHOULD show those as locked, it it doesn't that is what I am fighting, what locks those variables?

Your attached 3MF file does load with Max Print Speed greyed out under `Auto Speed (advanced)` - but that appears to be because you have set `Volumetric Speed` to zero. If you set a nonzero `Volumetric Speed`, it will unlock `Max Speed`.

This makes sense because auto-speed requires you to specify your max volumetric flow rate, and `Max Speed` caps the computed speed value. It makes no sense to set a Max Speed if you have not set a valid flow rate.

Regarding the `Default Speed` setting under `Speed for Print Moves`, there is an explanation of that in the pop-up help balloon:

> This setting doesn't do anything by itself, and so is deactivated unless a speed depends on it (a % from the left column) .

So in other words, if this is greyed out, it is telling you it is having no effect on anything.

inspectionsbybob, 2023-01-14:
Hum...
I think some things behavior has changed.
I didn't use to have to put in a Volumetric flow before,  which was good because I think in mm/sec not mm^3/sec

I thought I used to be able to leave them both Zero, and they defaulted to the Printer limits if there were any...

I will look up what the extruder can do with a 40W heater and take 75% of that as a good Max value for flow...

NameOfTheDragon, 2023-01-21:
> Hum... I think some things behavior has changed. I didn't use to have to put in a Volumetric flow before, which was good because I think in mm/sec not mm^3/sec
>
> I thought I used to be able to leave [...]

To be clear, both of the settings in the `Auto Speed (advanced)` box relate to automatic speed computation based on volumetric flow rate. These two are inextricably coupled since you cannot use auto-speed without setting a maximum volumetric flow rate. Therefore, `Max print speed` from this section is irrelevant if you are not using flow-rate based auto-speed. `Max print speed` only sets the maximum value that will be calculated. The printer can still go faster than this speed if you have set a faster speed explicitly in the `Speed for print moves` section. I guess there's a reason these items are in the _Advanced_ category ;-)

In order to use auto-speed, you MUST set the `Volumetric speed` and the `Max print speed`. Note: `Volumetric speed` can also be overridden per filament in filament settings. Under `Speed for print moves` you can then enter zero for any speed setting and SS will calculate that speed for you based on your volumetric flow rate.

This actually makes a lot of sense to do because your flow rate is one of the two fundamental constraints on print speed (see also: https://github.com/Timtek-Systems/Adventures-In-3D/blob/main/On%20Speed%20vs%20Acceleration/Why%20Speed%20is%20the%20Wrong%20Thing%20to%20Focus%20On.md ).

If you don't want to use auto-speed, then set `Volumetric speed` to zero and set nonzero speeds for all the settings under `Speeds for print moves`.

Note: It is best to set explicit speeds for `Gap fill speed/maximum speed` and `Thin walls` otherwise auto-speed goes a bit haywire for reasons I can't explain.

inspectionsbybob, 2023-01-22:
Hum...
My extruder (Biqu H2) can flow a LOT more than the nozzle can put down
without problems, ie in excess of 600mm^3.

So, if I get it right the Hierarchy of flow controls goes...
Filament override max flow
Filament override max mm/sec
Print settings/max volumetric speed flow
Print setting/speed max print speed
print settings Default Print Speed
Speed for individual print moves.

Question though... the DEFAULT  -FILAMENT- setting, filament, print speed
override for both Max speed and max volumetric is ZERO... where I have left
it at forever.
I tried to control my print speed by print/settings "Default Speed"  with
print settings, autospeed (advanced) Volumetric set to default of, again,
ZERO.  Max print speed is greyed out.  Am I missing something?

Most of the time my extruder can (greatly) outrun cooling and filament
"tearing" or not coming out of the nozzle laminar... this is often the max
of the recommended mm/sec listed on the filament roll its self.

So, I guess my real question is, where is the best place to control speed
and what to set all the DEFAULTS to?

-- 
-----------------------------------------------------------------------------------
Need a quick answer? Call our booking line at (301) 208-8289, or book your
inspection on our website 24/7.
Don't wait for a response as someone else may get the spot you want... call
Today!
-----------------------------------------------------------------------------------
Bob Sisson, ACI, BVI
Inspections by Bob, LLC
www.inspectionsbybob.com
***@***.***
MD Lic#29666; ASHI Member #212016

neophyl, 2023-01-22:
If you want to use autospeed then you have to set a max volumetric somewhere.  The 2 places (filament and print settings) I'm pretty sure it uses the lowest of whatever is set in the 2, its a limit so uses the lowest of the configured values.  Its not a case of one over riding the other, and zero means don't use.