supermerill/SuperSlicer #3570: Setting not saved

wolfgangmauer, 2023-01-12:
### What happened?

The setting "Top fill Pattern" is not saved.
[image]

after reopen the project

[image]

### Project file & How to reproduce

On an Object ->Add Settings -> Infill -> choose "Top fill Pattern"
Save Project -> Quit
Start new -> Open Recent projects ...

Setting "Top fill Pattern" is gone

### Version

2.4.58.5

### Operating system

windows 10

### Printer model

Voron 2.4 R2

neophyl, 2023-01-12:
Can you please please search for duplicate issues before opening new ones 
https://github.com/supermerill/SuperSlicer/issues/3409  

I used your own post title as a search term too.

wolfgangmauer, 2023-01-12:
I am so sorry