supermerill/SuperSlicer #3571: "Avoid Crossing" doesn't respect Ironed Surfaces

inspectionsbybob, 2023-01-13:
### Version
2.4.58.5

### Operating system type + version
Win10

### Behavior
After Ironing, PrintHead (sometimes) traverses the finished surface leaving "Scratches"

I have "Avoid Crossing Perimeters" turned on, but it still crosses and leaves these scratches after spending HOURS Ironing.

It doesn't happen with EVERY instance; it depends on where it is going NEXT. If the next object/layer to be printed is back over the finished surface, SCRATCH.  If next to it, it is left pristine.

_Is this a new feature request?_
NO

[16 Leafs.zip](https://github.com/supermerill/SuperSlicer/files/10413881/16.Leafs.zip)

#### Project File (.3MF) where problem occurs
_*Upload a Project File* (.3MF) `Plater -> Save Project as...`_
_You have to zip it (or rename it to .zip) to be able to drag& drop it here_

robustini, 2023-01-19:
Unfortunately Prusa Slicer also does this, it is a feature just done wrong.
In my case I would like the nozzle to always move inside the print, but instead this is the result.

[image]

inspectionsbybob, 2023-01-22:
I would like to see the printhead GO AROUND finished objects rather than taking the shortest, direct/straight, path to the next print location. Isn't that what "avoid printed objects" supposed to do?