supermerill/SuperSlicer #3574: Uncaught exception: boost::filesystem::path codecvt to wstring: error

Milas22, 2023-01-14:
### What happened?

Downloaded the nightly build on my windows Huawei notebook. Software does not start. Just shows error at the console. 

C:\Users\xxx\Desktop\SuperSlicer_2.5.59.2>superslicer_console.exe 
[0x000021a0] [trace]   Initializing StaticPrintConfigs
[0x000021a0] [error]   Unable to parse version of AMD driver.
[0x000021a0] [error]   Uncaught exception: boost::filesystem::path codecvt to wstring: error

### Project file & How to reproduce

At startup

### Version

2.5.59.2

### Operating system

Windows 11

WilQ18, 2024-03-18:
Same problem, Windows 11. My Windows account name contains Polish letter "ł" (capital - "Ł"). Version of SuperSlicer is 2.5.59.8.

[image]

[image]

Gott-deGreese, 2024-05-02:
You can create a user using the Latin alphabet, exclude him from all groups so that it is not visible on the welcome screen and run SuperSlicer using runas. I did it.
The problem is actually in the encoding of the username.

Felix14-v2, 2024-11-30:
> You can create a user using the Latin alphabet, exclude him from all groups so that it is not visible on the welcome screen and run SuperSlicer using runas. I did it.

That's not a good solution. You still can't open models from your main user (access denied), can't load profiles, can't use drag-n-drop etc.